rm-hull/luma.examples #3: Catch import errors in demo_opts

thijstriemstra, 2017-01-15:
Got the following traceback:

```python
luma.examples $ python examples/3d_box.py -i spi
Traceback (most recent call last):
  File "examples/3d_box.py", line 16, in <module>
    from demo_opts import device
  File "/home/pi/projects/pi-test/luma.examples/examples/demo_opts.py", line 12, in <module>
    import luma.lcd.device
ImportError: No module named 'luma.lcd'
```

I was trying to test an OLED example and it shouldn't complain when an optional library cannot be found.

rm-hull, 2017-01-15:
Yep, had considered that, and contemplated changing the import to:

```python
try:
    import luma.lcd.device
except ImportError:
    pass

try:
    import luma.oled.device
except ImportError:
    pass
```

Which would get you past the issue, but would fail a bit later on if you picked `-d pcd8544` 

To be fair, the README does say:

>     $ sudo -H pip install --upgrade luma.oled luma.lcd

:-p

thijstriemstra, 2017-01-15:
If the imports are moved instead of caught it'll make more sense.

rm-hull, 2017-01-15:
Yes, much better idea :+1:

commit b4a31ade (2017-01-15): Catch import errors in demo_opts - fixes #3
closed 2017-01-15 by rm-hull