rm-hull/luma.examples #4: PiCamera example

thijstriemstra, 2017-01-16:
picamera stream -> luma -> oled/lcd etc..

rm-hull, 2017-01-16:
@thijstriemstra - have a look at what @bjerrep is doing with gstreamer: https://github.com/bjerrep/ssd1306#television - amazing stuff!

thijstriemstra, 2017-01-16:
Wow that's awesome!

rm-hull, 2017-01-16:
Also, work is coming on with the led_matrix revamp, including a new emulator transformer:

[image]

```
$ cat led_matrix.conf 
--display=gifanim
--transform=led_matrix
--height=8
--width=40
--scale=16
--mode=1
--max-frames=375
--duration=0.05

$ python examples/tweet_scroll.py -f led_matrix.conf 
```

thijstriemstra, 2017-01-29:
@rm-hull Got it to work and it's magical: https://youtu.be/A9psiIqRNnc

Going to create a branch with these new examples now.

thijstriemstra, 2017-01-29:
Branch is at `picamera-4`. Do you have a camera to test with @rm-hull?

rm-hull, 2017-01-29:
Awesome stuff!
would be interesting to try and integrate it in with @bjerrep's gstreamer television code.

Haven't got a Pi camera though... 😭

closed 2017-01-29 by thijstriemstra

thijstriemstra, 2017-01-30:
Ah too bad, they're only 5 bucks though ;) http://www.ebay.com/itm/191823838867?_trksid=p2055119.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT

commit bea50f5b (2017-01-29): Merge pull request #9 from rm-hull/picamera-4

add picamera examples (fixes #4)