rm-hull/luma.examples #5: Don't hardcode display support

thijstriemstra, 2017-01-17:
The arguments in the example options parser should be dynamically populated. Basically every lib should advertise what display modes/interfaces are available.

thijstriemstra, 2017-01-25:
This also requires some sort of API coming from luma.core I imagine. Any library or simple plugin system we can use here? I have some of these escpos printers and I'm thinking of creating luma.escpos just for fun, that basically provides an emulator that prints every frame directly to a thermal printer. So I'd create such a package and then luma.examples should somehow know (like django) that there's an escpos emulator available (or a different oled display for example) when a certain package is installed. Thoughts?

rm-hull, 2017-01-25:
i guess the first place to start looking might be the [inspect](https://docs.python.org/2/library/inspect.html) module or [pyclbr](https://docs.python.org/2/library/pyclbr.html) possibly?

Alternatively recursively check and descend into `list(pkgutil.walk_packages(luma.__path__))` importing packages as you go

rm-hull, 2017-03-05:
Fixed in #18

closed 2017-03-05 by rm-hull