rm-hull/luma.examples #7: Noob guide

fullphat, 2017-01-23:
I have everything installed, but I'm struggling with the command-line options and receive this error each time:

Traceback (most recent call last):
  File "demo.py", line 16, in <module>
    from demo_opts import device
  File "/home/pi/luma.examples/examples/demo_opts.py", line 12, in <module>
    import luma.core.emulator
ImportError: No module named luma.core.emulator

Any chance of a walkthrough using the PCD8544 LCD as an example?

Thanks

rm-hull, 2017-01-23:
* Is this on Kali Linux or Raspian?
* Is this a Raspberry Pi or some other small board computer?
* How did you install the libs, through pip or `python setup.py install` in each of the projects
* What version of pip?
* What packages are installed? (`pip list`)

fullphat, 2017-01-23:
Hi, sorry...

Raspbian Jesse
Raspberry Pi 2
Used pip to install

pi@study:~ $ pip -V
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)

pi@study:~ $ pip list
argparse (1.2.1)
bitarray (0.8.1)
chardet (2.3.0)
colorama (0.3.2)
gpiozero (1.3.1)
html5lib (0.999)
luma.core (0.1.12-)
luma.lcd (0.2.0)
luma.led-matrix (0.3.3)
luma.oled (2.1.0)
lxkeymap (0.1)
mcpi (0.1.1)
ndg-httpsclient (0.3.2)
numpy (1.8.2)
olefile (0.44)
pcd8544 (0.1.0)
picamera (1.12)
pifacecommon (4.2.1)
pifacedigitalio (3.1.0)
pigpio (1.30)
Pillow (4.0.0)
pip (1.5.6)
pyasn1 (0.1.7)
pyephem (3.7.6.0)
pyfiglet (0.7.5)
pygame (1.9.2a0)
pygobject (3.14.0)
pyOpenSSL (0.13.1)
pyserial (2.6)
python-apt (0.9.3.12)
requests (2.4.3)
rpi-ws281x (2.0.3)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
sense-hat (2.2.0)
setuptools (5.5.1)
six (1.8.0)
smbus2 (0.1.4)
spidev (3.3)
unicornhat (2.1.2)
urllib3 (1.9.1)
wheel (0.24.0)
wiringpi (2.32.1)
wiringpi2 (2.32.3)
wsgiref (0.1.2)

rm-hull, 2017-01-23:
Ok, two things stand out
* the default version of pip is quite old (I'm on 9.0.1 here). 
* there's something a bit odd in `luma.core (0.1.12-)` ...there shouldn't be a dash ('-') after the 0.1.12; let me check how the release was built

Can you try:

```
$ sudo pip uninstall luma.core luma.lcd luma.oled luma.led_matrix
```

Answer yes if if asks you to proceed

Then:

```
$ sudo pip install --upgrade pip
$ pip --version
```

To verify you are on the latest version, then do:

```
$ sudo pip install luma.led_matrix luma.lcd luma.oled
$ pip list
```

And paste the results again, and then try the examples again

fullphat, 2017-01-23:
Ok, done!  pip looks like it's upgraded ok:

`Successfully installed pip
Cleaning up...
pi@study:~ $ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)`

And here's the results of pip list:

`pi@study:~ $ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
bitarray (0.8.1)
chardet (2.3.0)
colorama (0.3.2)
gpiozero (1.3.1)
html5lib (0.999)
luma.core (0.1.12)
luma.lcd (0.2.0)
luma.led-matrix (0.3.3)
luma.oled (2.1.0)
lxkeymap (0.1)
mcpi (0.1.1)
ndg-httpsclient (0.3.2)
numpy (1.8.2)
olefile (0.44)
pcd8544 (0.1.0)
picamera (1.12)
pifacecommon (4.2.1)
pifacedigitalio (3.1.0)
pigpio (1.30)
Pillow (4.0.0)
pip (9.0.1)
pyasn1 (0.1.7)
pyephem (3.7.6.0)
pyfiglet (0.7.5)
pygame (1.9.2a0)
pygobject (3.14.0)
pyOpenSSL (0.13.1)
pyserial (2.6)
python-apt (0.9.3.12)
requests (2.4.3)
rpi-ws281x (2.0.3)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
sense-hat (2.2.0)
setuptools (5.5.1)
six (1.8.0)
smbus2 (0.1.4)
spidev (3.3)
unicornhat (2.1.2)
urllib3 (1.9.1)
wheel (0.24.0)
wiringpi (2.32.1)
wiringpi2 (2.32.3)`

Different error this time! 

```
pi@study:~/luma.examples/examples $ ./demo.py --display pcd8544
usage: demo.py [-h] [--config CONFIG]
               [--display {ssd1306,ssd1325,ssd1331,sh1106,pcd8544,max7219,capture,pygame,gifanim}]
               [--width WIDTH] [--height HEIGHT] [--rotate {0,1,2,3}]
               [--interface {i2c,spi}] [--i2c-port I2C_PORT]
               [--i2c-address I2C_ADDRESS] [--spi-port SPI_PORT]
               [--spi-device SPI_DEVICE] [--spi-bus-speed SPI_BUS_SPEED]
               [--bcm-data-command BCM_DATA_COMMAND] [--bcm-reset BCM_RESET]
               [--bcm-backlight BCM_BACKLIGHT]
               [--transform {none,identity,scale2x,smoothscale,led_matrix,seven_segment}]
               [--scale SCALE] [--mode {1,RGB,RGBA}] [--duration DURATION]
               [--loop LOOP] [--max-frames MAX_FRAMES]
demo.py: error: SPI device not found
```

fullphat, 2017-01-23:
So, after enabling SPI (oops...) and rebooting, the demos now work, although I can't see anything displayed.

rm-hull, 2017-01-23:
If you've followed http://binerry.de/post/25787954149/pcd8544-library-for-raspberry-pi and wired it up like that then then this would explain why it doesn't work with this library.

Follow the wiring diagram in http://luma-lcd.readthedocs.io/en/latest/hardware.html#pre-requisites (_Pay particular heed of the warning in the text though, as the layout of the pins can differ from device to device; if in doubt, take a photo of the breakout board and drag the .jpg to the input box of this issue -- it will upload it for you_), then create a file called **pcd8544.conf** with the content as follows:

```
--display=pcd8544
--interface=spi
--spi-bus-speed=32000000
--bcm-reset=24
--bcm-data-command=23
--bcm-backlight=18
--width=84
--height=48
```

Then invoke an example as follows:

    $ examples/demo.py -f pcd8544.conf

fullphat, 2017-01-23:
Ah, sorry - I should explain that I've not actually wired it up myself - it came as a sealed unit (https://www.amazon.co.uk/SunFounder-Raspberry-PCD8544-Backlight-Compatible/dp/B01GPV7M1W/ref=sr_1_1?ie=UTF8&qid=1485198921&sr=8-1&keywords=pcd8544)

Still no dice using that conf file though I'm afraid...

fullphat, 2017-01-23:
Further update.  Found this page: http://www.sunfounder.com/wiki/index.php?title=Raspberry_Pi_5110_Mini_LCD_84*48_PCD8544_Usage which obviously covers the specific board/setup I have.  I followed the instructions and it displayed the test image correctly.  The text states this board doesn't use SPI at all, so won't I be able to use your library?

rm-hull, 2017-01-23:
So, unfortunately that unit is not going to work with this library as-is*, looking at the back (as below) it is hardwired to use GP0-4, none of which are SPI ports. It _appears_ to be using the I2C pins, but probably configured as GPIO pins.

[image]

The binerry software looks like it does software bit-banging to emulate SPI protocol.

_\* - it would be possible to write a software SPI bitbang (along the lines of https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/SPI.py#L150), but it would be slow in pure python - I can't promise anything, but feel free to raise a request in the issues on **luma.core**_

fullphat, 2017-01-23:
Understood - thanks for your efforts.  I may try to make some progress using the Adafruit library at least so I can see how far I can get with it.  :)

closed 2017-01-23 by fullphat

thijstriemstra, 2017-01-23:
I haven't tested this library yet but it should be good to go @rm-hull (and otherwise I'll file a ticket)?