rm-hull/luma.examples #11: Some examples not working on Rasp Pi 3 with SH1106 128x64

sheridat, 2017-02-11:
Hi, Just installed luma.core and luma.oled and now working through the luma.examples. Most of them work no problem and are fantastic!!!.  A few have issues - I  don't have the experience to determine why and whether it is my pi. 

The examples that are not working are:
1. tv_snow.py
2. terminal.py
3. carousel.py
4. welcome.py

I changed demo_opts.py to default to sh1106.
Below I have shown what happens when I run these 4.

1. tv_snow.py

 python tv_snow.py
Traceback (most recent call last):
  File "tv_snow.py", line 49, in <module>
    main()
  File "tv_snow.py", line 38, in main
    draw.multiline_text(offset, "Please do\nnot adjust\nyour set", fill="white", align="center", spacing=-1)
AttributeError: ImageDraw instance has no attribute 'multiline_text'

2. terminal py

python terminal.py or (python terminal.py --height 64)
Traceback (most recent call last):
  File "terminal.py", line 82, in <module>
    main()
  File "terminal.py", line 32, in main
    term.println("Uses any font to output text using a number of different print methods.")
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 219, in println
    self.puts(text)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 245, in puts
    self.putch(char, flush=self.animate)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 256, in putch
    self.newline()
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 292, in newline
    self._canvas.rectangle((0, copy.height, self._device.width, self._device.height), fill=self.bgcolor)
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 608, in __getattr__
    raise AttributeError(name)
AttributeError: height

3. carousel.py

python carousel.py
Traceback (most recent call last):
  File "carousel.py", line 19, in <module>
    from hotspot import memory, uptime, cpu_load, clock, network, disk
  File "/home/pi/luma.examples/examples/hotspot/memory.py", line 7, in <module>
    from hotspot.common import bytes2human, right_text, title_text, tiny_font
  File "/home/pi/luma.examples/examples/hotspot/common.py", line 9, in <module>
    tiny_font = ImageFont.truetype("examples/fonts/FreePixel.ttf", 10)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 240, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 137, in __init__
    self.font = core.getfont(font, size, index, encoding)
IOError: cannot open resource

4. welcome.py

python welcome.py
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'
ImageDraw instance has no attribute 'multiline_textsize'

Can you advise.

rm-hull, 2017-02-11:
Hello,

it sounds like you may have an older version of [pillow](https://pillow.readthedocs.io/en/4.0.x/) installed.

From the command line on your RPi, can you try:

    $ pip list | grep -i pillow

If it reports a version other than 4.0.0, this is the most likely cause. You should be able to upgrade with the following:

    $ sudo -H pip install --upgrade pillow

Then once that completes, try the `pip list | grep -i pillow` command again to confirm that 4.0.0 was installed, and try the demos again.

Let me know how it goes,

sheridat, 2017-02-11:
Yes - my Pillow is 2.6.1 
I attempted to update by "sudo -H pip install --upgrade pillow"
It downloaded Pillow-4.0.0.tar.gz however the install failed 
Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-GKS_7y/pillow/setup.py", line 791, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       http://pillow.readthedocs.io/en/latest/installation.html

and a bunchload more.

I tried to uninstall pillow but pip will not allow it as below. 
pi@raspberrypi:~ $ sudo pip uninstall Pillow
Not uninstalling Pillow at /usr/lib/python2.7/dist-packages, owned by OS

I'll go and have a look at the Pillow instillation docs. 

Thank You for pointing me in the right direction !

rm-hull, 2017-02-11:
Ah, did you follow the instructions in the README, specifically:

    $ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev

Then try upgrading pillow via pip?

rm-hull, 2017-02-11:
The pip-installed version _should_ be able to co-exist with the OS-installed version, but you can always uninistall it with:

    $ sudo apt-get purge python-pil

sheridat, 2017-02-11:
You've got me - I hadn't followed the instructions. I have followed then now and I have been able to upgrade to Pillow 4. Thank You.

I retested and now tv,snow, terminal and welcome all work flawlessly.

carousel.py still fails in the same way
Cpi@raspberrypi:~/luma.examples/examples $ python carousel.py
Traceback (most recent call last):
  File "carousel.py", line 19, in <module>
    from hotspot import memory, uptime, cpu_load, clock, network, disk
  File "/home/pi/luma.examples/examples/hotspot/memory.py", line 7, in <module>
    from hotspot.common import bytes2human, right_text, title_text, tiny_font
  File "/home/pi/luma.examples/examples/hotspot/common.py", line 9, in <module>
    tiny_font = ImageFont.truetype("examples/fonts/FreePixel.ttf", 10)
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFont.py", line 238, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFont.py", line 127, in __init__
    self.font = core.getfont(font, size, index, encoding)
IOError: cannot open resource

closed 2017-02-11 by sheridat

rm-hull, 2017-02-11:
ok, so the carousel has a relative path that is not quite right if you run it from within the examples directory --  I'll fix that up, but if in the meantime, you go up a directory, and try this from the project root:

    $ python example/carousel.py 

then it should work as expected

rm-hull, 2017-02-11:
Carousel example now fixed (ie. it can be run from within the examples directory) if you pull from master.

Note, you will need to install **psutil** to make it work though:

    $ sudo pip install psutil

See here: https://github.com/rm-hull/luma.examples/blob/master/examples/carousel.py#L13-L15

sheridat, 2017-02-11:
Got It. Thank You very much for helping me. 
Must admit that Carousel is my favourite. That's what my PI is going to be displaying from now on.
Cheers