rm-hull/luma.examples #23: display neopixel not working

thijstriemstra, 2017-03-09:
`test_get_device_led_matrix_all ` also fails on Rpi2:

```
_____________________________________________________ test_get_device_led_matrix_all _____________________________________________________

capsys = <_pytest.capture.CaptureFixture instance at 0x7491c440>

    def test_get_device_led_matrix_all(capsys):
        """
        Load supported led_matrix devices one by one.
        """
        for display in display_types.get('led_matrix'):
            try:
>               get_device(['--display', display])

tests/test_demo_opts.py:129: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
examples/demo_opts.py:157: in get_device
    device = Device(Serial(), **vars(args))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <luma.led_matrix.device.neopixel object at 0x74920ed0>, dma_interface = <luma.core.serial.spi object at 0x74920e70>, width = 128
height = 64, cascaded = None, rotate = 0, mapping = None
kwargs = {'bcm_backlight': 18, 'bcm_data_command': 24, 'bcm_reset': 25, 'bgr': False, ...}

    def __init__(self, dma_interface=None, width=8, height=4, cascaded=None,
                 rotate=0, mapping=None, **kwargs):
        super(neopixel, self).__init__(const=None, serial_interface=noop)
    
        # Derive (override) the width and height if a cascaded param supplied
        if cascaded is not None:
            width = cascaded
            height = 1
    
        self.cascaded = width * height
        self.capabilities(width, height, rotate, mode="RGB")
        self._mapping = list(mapping or range(self.cascaded))
        assert(self.cascaded == len(self._mapping))
        self._ws2812 = dma_interface or self.__ws2812__()
>       self._ws2812.init(width * height)
E       AttributeError: 'spi' object has no attribute 'init'

.tox/py27/local/lib/python2.7/site-packages/luma/led_matrix/device.py:197: AttributeError
```

On rpi2:

```python
$ python examples/3d_box.py --display neopixel
Traceback (most recent call last):
  File "examples/3d_box.py", line 129, in <module>
    device = get_device()
  File "/home/pi/projects/pi-test/luma.examples/examples/demo_opts.py", line 157, in get_device
    device = Device(Serial(), **vars(args))
  File "/home/pi/projects/pi-test/luma.led_matrix/luma/led_matrix/device.py", line 197, in __init__
    self._ws2812.init(width * height)
AttributeError: 'spi' object has no attribute 'init'
```

On Ubuntu machine I get:

```
$ python examples/3d_box.py --display neopixel
usage: 3d_box.py [-h] [--config CONFIG]
                 [--display {ssd1306,ssd1322,ssd1325,ssd1331,sh1106,pcd8544,st7735,max7219,neopixel,capture,gifanim,pygame}]
                 [--width WIDTH] [--height HEIGHT] [--rotate {0,1,2,3}]
                 [--interface {i2c,spi}] [--i2c-port I2C_PORT]
                 [--i2c-address I2C_ADDRESS] [--spi-port SPI_PORT]
                 [--spi-device SPI_DEVICE] [--spi-bus-speed SPI_BUS_SPEED]
                 [--bcm-data-command BCM_DATA_COMMAND] [--bcm-reset BCM_RESET]
                 [--bcm-backlight BCM_BACKLIGHT]
                 [--block-orientation {horizontal,vertical}]
                 [--mode {1,RGB,RGBA}]
                 [--framebuffer {diff_to_previous,full_frame}]
                 [--bgr {True,False}]
                 [--transform {identity,led_matrix,none,scale2x,seven_segment,smoothscale}]
                 [--scale SCALE] [--duration DURATION] [--loop LOOP]
                 [--max-frames MAX_FRAMES]
3d_box.py: error: SPI device not found
```

thijstriemstra, 2017-03-13:
@rm-hull any idea?

rm-hull, 2017-03-13:
I _think_ it can be solved with (at least for the RPi):

```diff
diff --git a/examples/demo_opts.py b/examples/demo_opts.py
index 5ab019d..a4a2986 100644
--- a/examples/demo_opts.py
+++ b/examples/demo_opts.py
@@ -154,7 +154,7 @@ def get_device(actual_args=None):
             from luma.core.serial import noop
             Device = getattr(luma.led_matrix.device, args.display)
             Serial = make_serial(args, gpio=noop()).spi
-            device = Device(Serial(), **vars(args))
+            device = Device(serial_interface=Serial(), **vars(args))
 
         elif args.display in display_types.get('emulator'):
             import luma.emulator.device
```

thijstriemstra, 2017-03-15:
Test succeeds on rpi so I consider this closed.

closed 2017-03-15 by thijstriemstra