rm-hull/luma.examples #24: Luma.OLED RPI3 ImportError: No module named luma....

flinthamm, 2017-03-09:
Sorry to be a pain but having spent an hour here and an hour there on this and followed through all the other people having similar issues, I still cannot get any of the examples to run on my RPI3 running the latest Raspian. 

I've remove, installed, pulled the latest examples down and have...
```
luma.core (0.5.4)
luma.oled (2.2.5)
```
...both installed. I'm also using pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

running python examples/demo.py gives:
`Traceback (most recent call last):
  File "examples/demo.py", line 17, in <module>
    from luma.core.render import canvas
ImportError: No module named luma.core.render
`
and as another example, running python examples/welcome.py gives:
`Traceback (most recent call last):
  File "examples/welcome.py", line 14, in <module>
    from luma.core.virtual import viewport, snapshot, range_overlap
ImportError: No module named luma.core.virtual
`

Any and all help would be much appreciated.

rm-hull, 2017-03-09:
Hi, what is the output of the following two commands:

    $ pip show luma.core

and, same for luma.oled:

    $ pip show luma.oled 

?

flinthamm, 2017-03-09:
Wow, thanks for your prompt response.
First command result
`Name: luma.core
Version: 0.5.4
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python3.4/dist-packages
Requires: smbus2, pillow, spidev, RPi.GPIO
`
Second...
`Name: luma.oled
Version: 2.2.5
Summary: A small library to drive an OLED device with either SSD1306, SSD1322, SSD1325, SSD1331 or SH1106 chipset
Home-page: https://github.com/rm-hull/luma.oled
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python3.4/dist-packages
Requires: luma.core
`
Both python 2.7 and 3.4 are installed on the RPI actually using a modified Raspian called Hassbian (i.e. Home Assistant) image. This is all being run under the user pi, with sudo when required/advised.

rm-hull, 2017-03-09:
Ok, so it looks like the packages were installed under python3, so you _should_ be able to run the demos with:

    $ python3 examples/welcome.py   # add opts/config for your device

Alternatively, if you specifically want to use python2, I would hedge a bet that you have a `pip2` on your system, so you could follow the install instructions again, substituting `pip2` wherever you read `pip`:

flinthamm, 2017-03-09:
Wow, was it really this simple! I could have sworn I tried this but clearly not! Thank you very much for your assistance and the great work your doing here of course too.  Now, fingers crossed I can get this SH1106 working with SPI.

flinthamm, 2017-03-09:
...and just to confirm. It works! Thanks again.

rm-hull, 2017-03-09:
:-)

closed 2017-03-09 by rm-hull

nishabe, 2017-12-28:
I am trying to run the examples and see the same issue described above.

pi@raspberrypi:~/luma.examples $ python examples/3d_box.py

```
Traceback (most recent call last):
  File "examples/3d_box.py", line 17, in <module>
    from demo_opts import get_device
  File "/home/pi/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error

```
Below are the different commands with the output:

pi@raspberrypi:~/luma.examples $ pip show luma.core
```
Name: luma.core
Version: 1.2.1
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python2.7/dist-packages/luma.core-1.2.1-py2.7.egg
Requires: pillow, smbus2, RPI.GPIO, spidev, monotonic
```
pi@raspberrypi:~/luma.examples $ pip show luma.oled

```
Name: luma.oled
Version: 2.3.1
Summary: A small library to drive an OLED device with either SSD1306, SSD1322, SSD1325, SSD1331, SSD1351 or SH1106 chipset
Home-page: https://github.com/rm-hull/luma.oled
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python2.7/dist-packages/luma.oled-2.3.1-py2.7.egg
Requires: luma.core
```
python --version:
`Python 2.7.13`
pip --version says:
`pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)`

Could you please help me understand the issue here?

rm-hull, 2017-12-28:
Hi @nishabe, you need to specify a display type on the command line, e.g.

    python examples/3d_box.py -d ssd1306

thijstriemstra, 2017-12-28:
> you need to specify a display type on the command line, e.g.

`ssd1306` is the default display type so that shouldn't be the issue I think..

nishabe, 2017-12-29:
I get:
```
pi@raspberrypi:~/luma/luma.examples $ python -c 'from luma.core import __version__'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named core
```

elpimous, 2017-12-29:
Did you use this link ?

https://luma-oled.readthedocs.io/en/latest/install.html

It seems that the core libs didn t install

nishabe, 2017-12-30:
Yes, I have.
These are the steps in the link:
https://luma-oled.readthedocs.io/en/latest/install.html
$ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev build-essential
$ sudo -H pip install --upgrade pip
$ sudo apt-get purge python-pip
$ sudo -H pip install --upgrade luma.oled

These are the instructions in github link:
https://github.com/rm-hull/luma.examples

$ sudo usermod -a -G i2c,spi,gpio pi
$ sudo apt install python-dev python-pip libfreetype6-dev libjpeg-dev build-essential
$ sudo apt install libsdl-dev libportmidi-dev libsdl-ttf2.0-dev libsdl-mixer1.2-dev libsdl-image1.2-dev
$ sudo -H pip install --upgrade pip setuptools
$ sudo apt purge python-pip

Can someone tell me why install pip only to purge it in the next step?
Because executing the final step ,  I get:
"sudo: pip: command not found"

rm-hull, 2017-12-30:
@nishabe: 

The pip installed on Jessie was quite old AFAICR, so we needed to bootstrap the latest pip and setuptools from pypi. Maybe that advice no longer applies on the latest Rasbian image.

You should just install the latest luma.oled using `sudo -H pip install --upgrade --force-reinstall --ignore-installed luma.oled` and if you could post the output of that command here

nishabe, 2017-12-30:
Thank you. It did fix it.
💯

surak, 2018-04-09:
A clean install does not work on Raspbian. You have to do the "sudo -H pip install --upgrade --force-reinstall --ignore-installed luma.oled" and it goes well. Maybe something wrong with the pip install process?

steelx, 2019-12-10:
- `sudo -H pip3 install --upgrade --force-reinstall --ignore-installed luma.oled`
- `sudo -H pip3 install -e .`
and then I did `python3 examples/3d_box.py --width 128 --height 128 --display sh1106`

and still, I get this error 
```
Traceback (most recent call last):
  File "examples/3d_box.py", line 17, in <module>
    from demo_opts import get_device
  File "/home/pi/projects/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error
ModuleNotFoundError: No module named 'luma'
```

please suggest

thijstriemstra, 2019-12-10:
what is the output of `sudo -H pip3 install --upgrade --force-reinstall --ignore-installed luma.oled`?

steelx, 2019-12-10:
@thijstriemstra  thanks for the reply.
Below is the output:
```bash
pi@raspberrypi:~ $ sudo -H pip3 install --upgrade --force-reinstall --ignore-installed luma.oled
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting luma.oled
  Using cached https://files.pythonhosted.org/packages/fd/70/b03e0cf2040113db644a220d546a9c42907c4884a898722312c83b0f5603/luma.oled-3.3.0-py2.py3-none-any.whl
Collecting luma.core>=1.12.0 (from luma.oled)
  Using cached https://files.pythonhosted.org/packages/a7/36/f10ac3202221c479bae89e3226e57dcbf31c08f4d9e5743f32e193147a22/luma.core-1.12.0-py2.py3-none-any.whl
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.12.0->luma.oled)
  Using cached https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp37-cp37m-linux_armv6l.whl
Collecting pillow>=4.0.0 (from luma.core>=1.12.0->luma.oled)
Collecting smbus2 (from luma.core>=1.12.0->luma.oled)
  Using cached https://www.piwheels.org/simple/smbus2/smbus2-0.3.0-py2.py3-none-any.whl
Collecting pyftdi; python_version >= "3.5" (from luma.core>=1.12.0->luma.oled)
  Using cached https://www.piwheels.org/simple/pyftdi/pyftdi-0.42.2-py35-none-any.whl
Collecting spidev; platform_system == "Linux" (from luma.core>=1.12.0->luma.oled)
  Using cached https://www.piwheels.org/simple/spidev/spidev-3.4-cp37-cp37m-linux_armv6l.whl
Collecting pyserial>=3.0 (from pyftdi; python_version >= "3.5"->luma.core>=1.12.0->luma.oled)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting pyusb>=1.0.0 (from pyftdi; python_version >= "3.5"->luma.core>=1.12.0->luma.oled)
  Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.0.2-py3-none-any.whl
Installing collected packages: RPI.GPIO, pillow, smbus2, pyserial, pyusb, pyftdi, spidev, luma.core, luma.oled
Successfully installed RPI.GPIO-0.7.0 luma.core-1.12.0 luma.oled-3.3.0 pillow-6.2.1 pyftdi-0.42.2 pyserial-3.4 pyusb-1.0.2 smbus2-0.3.0 spidev-3.4
```

thijstriemstra, 2019-12-10:
**sudo** `python3 examples/3d_box.py --width 128 --height 128 --display sh1106`?

steelx, 2019-12-10:
i get below error:
```python
pi@raspberrypi:~/projects/luma.examples $ python3 examples/3d_box.py --width 128 --height 128 --display sh1106
Traceback (most recent call last):
  File "examples/3d_box.py", line 17, in <module>
    from demo_opts import get_device
  File "/home/pi/projects/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error
ModuleNotFoundError: No module named 'luma'
```

thijstriemstra, 2019-12-10:
yea, try that command with `sudo`

steelx, 2019-12-10:
omg. worked with `sudo` why was this ? Sorry, I just installed python first time. Im trying to get this running for GoLang after

thijstriemstra, 2019-12-10:
it's because you installed luma using `sudo pip3`. Take a look at `virtualenvwrapper` utility to create a virtual environment where you can install luma without sudo.

steelx, 2019-12-10:
thanks again :) @thijstriemstra