rm-hull/luma.examples #26: fontawesome icons example

thijstriemstra, 2017-03-15:
Would be cool to have a [fontawesome](http://fontawesome.io/) example that demonstrates how to render their icons (on an oled screen).

thijstriemstra, 2017-03-15:
was about to start with this ha, but I'll pick up another issue. Looking forward to this example!

I would like to have something that displays all of the icons one by one, with a few seconds delay, with a large icon centered on the screen and the icon name below it or something?

rm-hull, 2017-03-15:
> I would like to have something that displays all of the icons one by one, 

check

> with a few seconds delay, 

check

> with a large icon centered on the screen 

check

> and the icon name below it 

ah...

😆

closed 2017-03-15 by rm-hull

thijstriemstra, 2017-03-15:
It's not mentioned in the readme.

commit 958736d2 (2017-03-15): Font awesome walkthrough - fixes #26