rm-hull/luma.examples #30: matplotlib example

thijstriemstra, 2017-03-17:
any matplotlib graph -> pil -> display

rm-hull, 2017-03-20:
What would be really cool would be to implement another pseudo-device in luma.core that: 
* can wrap an existing luma display device
* subclasses [MovieWriter](http://matplotlib.org/api/_as_gen/matplotlib.animation.MovieWriter.html#matplotlib.animation.MovieWriter)

This would pump frames to the underlying device, enabling all the existing [matplotlib animation demos](http://matplotlib.org/api/animation_api.html#examples) to work more-or-less unchanged.

thijstriemstra, 2017-11-20:
i guess this would be implemented in luma.emulator?

rm-hull, 2017-11-20:
I did have a kick around with this a few weeks (months?) back, but never committed anything. I reckon assume luma.core might be the right destination - it would be a wrapper much like the terminal is.