rm-hull/luma.examples #33: Font rendering scripts fail to execute.

somu1795, 2017-03-22:
The scripts which use fonts from the "fonts" folder included in examples, fail to execute.
It raises IOError - unknown file format.

It seems, imagefont is unable to load fonts  from anywhere else except the default folders, i.e -
 "/usr/share/fonts/truetype/freefont/"
After copying the individual fonts to the above mentioned folder, the script works flawlessly, also the provided "code2000.ttf" font is corrupt.

thijstriemstra, 2017-03-22:
@somu1795 what operating system, python version, pygame version etc?

somu1795, 2017-03-22:
OS: raspbian latest
Python - 2.7.9
Pygame - 1.9.3
I updated the issue with the error included.

thijstriemstra, 2017-03-22:
can you show `ls -l` for corrupt and correct code2000.ttf? and where did you get a fresh copy?

somu1795, 2017-03-22:
```
$ ls -l
total 6224
-rwxr-xr-x 1 root root   49432 Mar 21 07:42 C&C Red Alert [INET].ttf
-rwxr-xr-x 1 root root 2686976 Mar 21 07:42 code2000_bak.ttf
-rwxr-xr-x 1 root root 3155104 Feb  3  2010 code2000.ttf
-rwxr-xr-x 1 root root   85484 Mar 21 07:42 creep.bdf
-rwxr-xr-x 1 root root  165548 Mar 21 07:42 fontawesome-webfont.ttf
-rwxr-xr-x 1 root root   64880 Mar 21 07:42 FreePixel.ttf
-rwxr-xr-x 1 root root   23280 Mar 21 07:42 miscfs_.ttf
-rwxr-xr-x 1 root root   22184 Mar 21 07:42 pixelmix.ttf
-rwxr-xr-x 1 root root   35656 Mar 21 07:42 ProggyTiny.ttf
-rwxr-xr-x 1 root root    2847 Mar 21 07:42 README [INET].txt
-rwxr-xr-x 1 root root    6656 Mar 21 07:42 tiny.ttf
-rwxr-xr-x 1 root root   55580 Mar 21 07:42 Volter__28Goldfish_29.ttf

```
The "code2000_bak.ttf" is the corrupt aka original one provided, and "code2000.ttf" is the new one downloaded from - [http://www.fonts2u.com/code2000.font](url)

thijstriemstra, 2017-03-22:
Hmm.. Ubuntu 16.04, latest git master for examples:

```
$ ls -l ../luma.examples/examples/fonts/
total 8700
-rw-rw-r-- 1 thijs thijs   49432 feb 27 23:59 C&C Red Alert [INET].ttf
-rw-rw-r-- 1 thijs thijs 8377000 feb 27 23:59 code2000.ttf
-rw-rw-r-- 1 thijs thijs   85484 feb 27 23:59 creep.bdf
-rw-rw-r-- 1 thijs thijs  165548 mrt 15 21:37 fontawesome-webfont.ttf
-rw-rw-r-- 1 thijs thijs   64880 feb 27 23:59 FreePixel.ttf
-rw-rw-r-- 1 thijs thijs   23280 feb 27 23:59 miscfs_.ttf
-rwxrwxr-x 1 thijs thijs   22184 feb 27 23:59 pixelmix.ttf
-rw-rw-r-- 1 thijs thijs   35656 feb 27 23:59 ProggyTiny.ttf
-rw-rw-r-- 1 thijs thijs    2847 feb 27 23:59 README [INET].txt
-rw-rw-r-- 1 thijs thijs    6656 feb 27 23:59 tiny.ttf
-rw-rw-r-- 1 thijs thijs   55580 feb 27 23:59 Volter__28Goldfish_29.ttf
```

thijstriemstra, 2017-03-22:
And git shows 7.9 MB: https://github.com/rm-hull/luma.examples/blob/master/examples/fonts/code2000.ttf so I think it got corrupted during your git checkout? Can you try to clone it again?

somu1795, 2017-03-22:
After cloning again, I do get the correct code2000.ttf 
```
$ ls -l
total 516
-rwxr-xr-x 1 root root   49432 Mar 22 20:04 C&C Red Alert [INET].ttf
-rwxr-xr-x 1 root root 8377000 Mar 22 20:04 code2000.ttf
-rwxr-xr-x 1 root root   85484 Mar 22 20:04 creep.bdf
-rwxr-xr-x 1 root root  165548 Mar 22 20:05 fontawesome-webfont.ttf
-rwxr-xr-x 1 root root   64880 Mar 22 20:04 FreePixel.ttf
-rwxr-xr-x 1 root root   23280 Mar 22 20:05 miscfs_.ttf
-rwxr-xr-x 1 root root   22184 Mar 22 20:05 pixelmix.ttf
-rwxr-xr-x 1 root root   35656 Mar 22 20:04 ProggyTiny.ttf
-rwxr-xr-x 1 root root    2847 Mar 22 20:04 README [INET].txt
-rwxr-xr-x 1 root root    6656 Mar 22 20:05 tiny.ttf
-rwxr-xr-x 1 root root   55580 Mar 22 20:04 Volter__28Goldfish_29.ttf

```

somu1795, 2017-03-22:
Thanks , it works after getting the fresh repository.
I downloaded the zip file instead of cloning this time , and everything works correctly.
Also I had to install the pygame through apt-get : 
`$ sudo apt-get install python-pygame`
Installing through the default installer script generates error, "sdl-config" not found.

thijstriemstra, 2017-03-22:
> Also in pcd8544.conf the default SPI speed is 32Mhz , but with this setting the display remains blank , it works upto a max of 8Mhz.
@rm-hull is this a bug?

rm-hull, 2017-03-22:
So its a trade off, 32MHz obviously transfers data quicker but is subject to cross chatter and interference - and the vagaries of the different breakout boards out there. It works for me, but other boards I have to crank it down to 16MHz to get it to work.

Maybe the conf file settings should be more conservative ... PR happily accepted @somu1795 :)

rm-hull, 2017-03-22:
re. the sdl-config, we're in the process of overhauling the docs into one place - at the moment its spread (and partially repeated) over 4 or 5 doc sites, but AFAIR following https://luma-emulator.readthedocs.io/en/latest/install.html#installing-from-pypi should be sufficient to install the latest pygame.

somu1795, 2017-03-22:
Aha! Thanks for that !

closed 2017-03-22 by somu1795