rm-hull/luma.examples #36: Installation instructions do not work

fiveseven808, 2017-03-26:
I followed:
$ sudo usermod -a -G i2c,spi,gpio pi
$ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev
$ sudo apt-get purge python-pip
$ sudo -H pip install --upgrade pip

However, the last command results in:
`sudo: pip: command not found`

rm-hull, 2017-03-26:
Oops - good spot: it looks like those last two commands were the wrong way round. Should be fixed now in the README.

You should be able to do:

    $ sudo apt-get install python-pip
    $ sudo -H pip install --upgrade pip
    $ sudo apt-purge python-pip

and then check it with:

    $ pip --version
    pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)

fiveseven808, 2017-03-26:
Thank you! I will try that!

fiveseven808, 2017-03-28:
It worked perfectly! Thanks!

closed 2017-03-28 by fiveseven808