rm-hull/luma.examples #37: Module requirements not installed for some examples

fiveseven808, 2017-03-26:
`pi@devpi ~/l/examples> ./carousel.py
Traceback (most recent call last):
  File "./carousel.py", line 18, in <module>
    import psutil
ImportError: No module named psutil
`

Simple fix, but maybe it an be included in the installer?

fiveseven808, 2017-03-26:
Actually... when installed I still get this error:

`pi@devpi ~/l/examples> ./carousel.py
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
Traceback (most recent call last):
  File "./carousel.py", line 110, in <module>
    main()
  File "./carousel.py", line 79, in main
    network_ifs = psutil.net_if_stats().keys()
AttributeError: 'ModuleWrapper' object has no attribute 'net_if_stats'
`

Any ideas?

rm-hull, 2017-03-26:
What version of psutil, I've got 5.0.1

    $ pip list | grep psutil
    psutil (5.0.1)

rm-hull, 2017-03-26:
The carousel does have some instructions in the script https://github.com/rm-hull/luma.examples/blob/master/examples/carousel.py#L13, but it's not obvious without looking at the code

fiveseven808, 2017-03-26:
```
$ pip list | grep psutil
psutil (2.1.1)
```
Looks like I ended up installing an older version. Any idea how to update?

rm-hull, 2017-03-26:
Try:

    $ sudo -H pip install --upgrade psutil

fiveseven808, 2017-03-26:
```
pi@devpi ~/HueBerry_SmartSwitch> sudo -H pip install --upgrade psutil
Downloading/unpacking psutil from https://pypi.python.org/packages/b8/47/c85fbcd23f40892db6ecc88782beb6ee66d22008c2f9821d777cb1984240/psutil-5.2.1.tar.gz#md5=ed731bcbb5737486a60807a1dc6183eb
  Downloading psutil-5.2.1.tar.gz (347kB): 347kB downloaded
  Running setup.py (path:/tmp/pip-build-Oey7J6/psutil/setup.py) egg_info for package psutil

    warning: manifest_maker: MANIFEST.in, line 14: 'recursive-include' expects <dir> <pattern1> <pattern2> ...

    warning: no previously-included files matching '*' found under directory 'docs/_build'
    warning: no previously-included files matching '*' found under directory '.ci'
Installing collected packages: psutil
  Found existing installation: psutil 2.1.1
    Not uninstalling psutil at /usr/lib/python2.7/dist-packages, owned by OS
  Running setup.py install for psutil
    building 'psutil._psutil_linux' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_VERSION=521 -DPSUTIL_LINUX=1 -I/usr/include/python2.7 -c psutil/_psutil_linux.c -o build/temp.linux-armv6l-2.7/psutil/_psutil_linux.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv6l-2.7/psutil/_psutil_linux.o -o build/lib.linux-armv6l-2.7/psutil/_psutil_linux.so
    building 'psutil._psutil_posix' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_VERSION=521 -DPSUTIL_LINUX=1 -I/usr/include/python2.7 -c psutil/_psutil_posix.c -o build/temp.linux-armv6l-2.7/psutil/_psutil_posix.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv6l-2.7/psutil/_psutil_posix.o -o build/lib.linux-armv6l-2.7/psutil/_psutil_posix.so

    warning: manifest_maker: MANIFEST.in, line 14: 'recursive-include' expects <dir> <pattern1> <pattern2> ...

    warning: no previously-included files matching '*' found under directory 'docs/_build'
    warning: no previously-included files matching '*' found under directory '.ci'
Successfully installed psutil
Cleaning up...
pi@devpi ~/HueBerry_SmartSwitch> pip list | grep psutil
psutil (2.1.1)
```

Looks like it doesn't like that I added it via apt-get... should I remove it and try again?

rm-hull, 2017-03-26:
Ah, yeah, I would always install from pip as the bundled versions from apt are often very old. If you purge the apt version, and retry with pip

fiveseven808, 2017-03-26:
Good to know! I did not realize this.

All is well now! Thank you!

closed 2017-03-26 by fiveseven808

thijstriemstra, 2017-03-26:
Again, we should throw an error for old psutil versions.