rm-hull/luma.examples #39: jetset_willy.py doesn't work?

fiveseven808, 2017-03-26:
```
pi@devpi ~/l/examples> ./jetset_willy.py
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
Traceback (most recent call last):
  File "./jetset_willy.py", line 100, in <module>
    main()
  File "./jetset_willy.py", line 54, in main
    sheet = spritesheet(**data)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/sprite_system.py", line 78, in __init__
    self.image = self.image.crop((regX, regY, self.image.width - regX, self.image.height - regY))
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 608, in __getattr__
    raise AttributeError(name)
AttributeError: width

```

rm-hull, 2017-03-26:
Sounds like you've got an older version of pillow installed. (4.0.0 is required). 

    $ pip list | grep -i Pillow
    Pillow (4.0.0)

Try upgrading to the latest with:

    $ sudo -H pip install --upgrade pillow

fiveseven808, 2017-03-26:
I had to removed the system installed one, but now it works great! Thank you very much!

closed 2017-03-26 by fiveseven808

thijstriemstra, 2017-03-26:
Maybe we can throw an error for older versions of Pillow.