rm-hull/luma.examples #40: pi_logo.py doesn't work?

fiveseven808, 2017-03-26:
```
pi@devpi ~/l/examples> ./pi_logo.py
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
Traceback (most recent call last):
  File "./pi_logo.py", line 36, in <module>
    main()
  File "./pi_logo.py", line 23, in main
    posn = ((device.width - logo.width) // 2, 0)
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 608, in __getattr__
    raise AttributeError(name)
AttributeError: width
```

I wonder what happened?

fiveseven808, 2017-03-26:
runner.py seems to experience the same issue? 

```
pi@devpi ~/l/examples> ./runner.py
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
Traceback (most recent call last):
  File "./runner.py", line 66, in <module>
    main()
  File "./runner.py", line 40, in main
    sheet = spritesheet(**data)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/sprite_system.py", line 78, in __init__
    self.image = self.image.crop((regX, regY, self.image.width - regX, self.image.height - regY))
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 608, in __getattr__
    raise AttributeError(name)
AttributeError: width

```

rm-hull, 2017-03-26:
Same issue as #39

closed 2017-03-26 by rm-hull

thijstriemstra, 2017-03-26:
Everything not executable as example should probably be moved to a dir/package