rm-hull/luma.examples #42: [Question] Support for 64x48 OLED Display?

csskevin, 2017-03-29:
Hello,
Would it be possible to add a support for a 64x48 OLED display. I do not have much experience here. Can I add it manual?
Thanks

thijstriemstra, 2017-03-29:
@csskevin do you have a link to the ebay/product page and/or datasheet of the device?

csskevin, 2017-03-29:
Bought it through amazon. The contributer link https://www.wemos.cc/product/oled-shield.html

thijstriemstra, 2017-03-29:
Have you tested it with the luma.examples? Can you try this command:

```
python examples/starfield.py --display ssd1306 --width 64 --height 48 --interface i2c
```

csskevin, 2017-03-29:
There I got following error:
starfield.py: error: Unsupported display mode: 64 x 48

rm-hull, 2017-03-29:
The good news is that it should work with a small fix, but presently won't work as-is. The device only supports certain modes - see https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L124-L129

To make it work we would need to find out the correct values for the multiplex and clock divider, and add an entry like:

```python
(64, 48): dict(multiplex=???, displayclockdiv=???, compins=???)
```
@csskevin I will have a look at the datasheet and see if we can suggest some values for you to try, how does that sound?

thijstriemstra, 2017-03-29:
ok. start a python console:

```
$ python
```

And type:

```
from luma.oled import device
device
```
This should print the path of the oled module, e.g.

```
$ python
Python 3.5.3 (default, Feb 18 2017, 01:21:45) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from luma.oled import device
>>> device
<module 'luma.oled.device' from '/home/me/projects/bar/src/luma.oled/luma/oled/device.py'>
>>> 
```

Now open `/home/me/projects/bar/src/luma.oled/luma/oled/device.py` with a text editor
and look for:

```
# Supported modes
        settings = {
            (128, 64): dict(multiplex=0x3F, displayclockdiv=0x80, compins=0x12),
            (128, 32): dict(multiplex=0x1F, displayclockdiv=0x80, compins=0x02),
            (96, 16): dict(multiplex=0x0F, displayclockdiv=0x60, compins=0x02)
}.get((width, height))
```
Copy this so it looks like:

```
# Supported modes
        settings = {
            (128, 64): dict(multiplex=0x3F, displayclockdiv=0x80, compins=0x12),
            (128, 32): dict(multiplex=0x1F, displayclockdiv=0x80, compins=0x02),
            (96, 16): dict(multiplex=0x0F, displayclockdiv=0x60, compins=0x02),
            (64, 48): dict(multiplex=0x0F, displayclockdiv=0x60, compins=0x02)
}.get((width, height))
```

And try again..

csskevin, 2017-03-29:
@rm-hull This would sound perfect. Thanks very much

@thijstriemstra Thank you, the  64 x 48 width and height works, but the dimensions are not working properly. I see just a part. (I see a part from right top)

thijstriemstra, 2017-03-29:
@csskevin can you post a picture?

csskevin, 2017-03-29:
Sorry fot the bad image quality. I started the demo.py, because on the startfield.py I see just dots.

[image]

thijstriemstra, 2017-03-29:
no worries, thanks for posting. Hopefully @rm-hull can find the correct values for that line and it should be working ok.

csskevin, 2017-03-29:
Thank you both very much and thanks for the fast help.

rm-hull, 2017-03-29:
Try: 

```python
(64, 48): dict(multiplex=0x2F, displayclockdiv=0x80, compins=0x12)
```

If you are going to test it, please use

    $ python examples/demo.py --display ssd1306 --width 64 --height 48 --interface i2c

It should look like this:

[image]

You can't tell from the screengrab but there should be white outline around the shapes

csskevin, 2017-03-29:
[image]

rm-hull, 2017-03-29:
Can you alter your local copy of https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L171 to read:

```python
self._const.COLUMNADDR, 0x20, 0x5F,
```

See what happens?

csskevin, 2017-03-29:
Thank you very very much, it works perfectly now

rm-hull, 2017-03-29:
Ok, so thats great. I'll roll out a fix for the general case, but I dont think we ever tested it for a 96x16 device - did you @thijstriemstra ? I suspect that never worked properly for the same reason the 64x48 didnt .. :-s

rm-hull, 2017-03-29:
See https://github.com/rm-hull/luma.oled/pull/144

thijstriemstra, 2017-03-29:
I couldn't find the ticket for 96x16 other than a really old one that relates to the previous codebase.

thijstriemstra, 2017-03-29:
Smallest OLED I tested: https://github.com/rm-hull/luma.oled/issues/23 (128x32)

rm-hull, 2017-03-30:
@csskevin can you upgrade to luma.oled 2.2.6 please and let me know if that fixes it for you? use `sudo -H pip install --upgrade luma.oled`

If so please close the ticket :-)

csskevin, 2017-03-30:
Thanks for your great support. It works perfectly :) 👍

closed 2017-03-30 by csskevin