rm-hull/luma.examples #43: Ignore unsupported dimensions error

thijstriemstra, 2017-03-29:
Instead of raising an error, e.g. https://github.com/rm-hull/luma.examples/issues/42#issuecomment-290218431, print a warning message instead. I think this will lead to more and better bug reports and enhancements.

rm-hull, 2017-04-12:
The dimension checks are there for a reason - if a size isn't explicitly mentioned, then it is unlikely to work. If someone has a device which isn't supported, I'd rather open up a dialog first to establish the need, rather than getting spurious bug reports.

thijstriemstra, 2017-04-12:
ok, what about adding an additional warning when an unsupported dimensions error is encountered, telling the user that it can be supported if they open a new ticket with the error message attached. I'd like to know what happens in the wild and support as many devices as possible.

closed 2017-07-03 by thijstriemstra