rm-hull/luma.examples #44: Install failure

sgjava, 2017-04-04:
I finally have SSD1306 SPI working on CHIP! I tried to install luma.examples:

- sudo apt-get install python-dev libsdl-image1.2-dev libsdl-mixer1.2-dev libsdl-ttf2.0-dev libsdl1.2-dev libsmpeg-dev python-numpy subversion libportmidi-dev ffmpeg libswscale-dev libavformat-dev libavcodec-dev
- git clone https://github.com/rm-hull/luma.examples.git
- cd luma.examples
- sudo -H pip install -e .

And get:

```
Obtaining file:///home/chip/luma.examples
    Complete output from command python setup.py egg_info:
    error in luma.examples setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/chip/luma.examples/

```

sgjava, 2017-04-04:
My bad, missed sudo -H pip install --upgrade pip setuptools

closed 2017-04-04 by sgjava

thijstriemstra, 2017-04-05:
Its a setuptools thing. Maybe we can catch that error and explain; upgrade to newer version.

sgjava, 2017-04-05:
Yea, I fixed that. I finally got an SPI display working on the CHIP (still a SSD1306) using CHIP_IO for GPIO. I ended up hacking luma.examples/examples/demo_opts.py:

- Add to imports:

import CHIP_IO.GPIO
from luma.core.serial import spi
from luma.oled.device import ssd1306

- Replace device = cmdline.create_device(args) with

serial = spi(port=32766, device=0, gpio=CHIP_IO.GPIO, bcm_DC="CSID1", bcm_RST="CSID0")
device = ssd1306(serial)

thijstriemstra, 2017-04-05:
Looks good. But use `gpio_DC` and `gpio_RST` instead, others are deprecated.

sgjava, 2017-04-05:
Crap, you told me that before :) I'm copying the same old code!

thijstriemstra, 2017-04-05:
You actually can pass all these args to demo_opts, except the gpio implementation. Somethig that should be added. Please open a ticket in core for enhancement to cmdline.py

thijstriemstra, 2017-04-05:
And PR of course ;)

thijstriemstra, 2017-04-05:
> I'm copying the same old code!

Any outdated code/examples; PRs welcome.