rm-hull/luma.examples #45: 'extras_require' must be a directory whose...

rendeko, 2017-04-09:
```
Obtaining file:///home/pi/luma.examples
    Complete output from command python setup.py egg_info:
    error in luma.examples setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/luma.examples/
```

This is from following the example on the front page. Examples run fine, just not the installation through this method, which I think may be one of my issues for a bigger issue I am having.

thijstriemstra, 2017-04-09:
We should catch this error or display better msg

thijstriemstra, 2017-04-12:
So basically this happens if your setuptools is not up to date and unable to handle the `extra_requires` stuff. The [default setuptools version on Raspbian Jessie](https://packages.debian.org/jessie/python-setuptools) is currently: 5.5.1. The feature we use was [introduced](http://setuptools.readthedocs.io/en/latest/history.html#a4) in setuptools 0.5a4. So it should already work? Hmm.

thijstriemstra, 2017-04-12:
The exception is also still present in todays setuptools: https://github.com/pypa/setuptools/blob/master/setuptools/dist.py#L137

thijstriemstra, 2017-04-12:
Well, someone already fixed it in a different project, let's use that: https://github.com/bigchaindb/bigchaindb/pull/279/files

thijstriemstra, 2017-04-12:
Here's the actual error:

```
$ pip install -e .
Obtaining file:///home/pi/projects/pi-test/luma.examples
    Complete output from command python setup.py egg_info:
    error in luma.examples setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/projects/pi-test/luma.examples/
```

thijstriemstra, 2017-04-12:
With PR #47:

```
$ pip install -e .
Obtaining file:///home/pi/projects/pi-test/luma.examples
    Complete output from command python setup.py egg_info:
    Your Python distribution comes with an incompatible version of `setuptools`. Please run:
     $ pip install --upgrade setuptools
    and then run this command again
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/projects/pi-test/luma.examples/
```

or should `sudo` be added to the command @rm-hull?

thijstriemstra, 2017-04-12:
The $ is also confusing.

thijstriemstra, 2017-04-12:
I suppose this check should be added to all project setup.py's?

closed 2017-04-12 by thijstriemstra

rm-hull, 2017-04-12:
> The default setuptools version on Raspbian Jessie is currently: 5.5.1. 

Is that true for python2 _and_ python3?

> The $ is also confusing.

I always think that `$` is a good indicator that that what follows is something that needs to keyed in at a bash prompt

> I suppose this check should be added to all project setup.py's?

Sounds like it

thijstriemstra, 2017-04-12:
> Is that true for python2 and python3?

no idea, probably.

thijstriemstra, 2017-04-12:
It looks like it's also not related to the ability to parse `extras_require` but the 'monotonic<python3.3' entry, which requires a much newer setuptools than 5.x I suspect.

cventour, 2018-03-12:
Hello all, 

I have updated pip and setuptools but still seeing the same error about extra_requires.

What info can I provide you to help you narrow it down ?

thijstriemstra, 2018-03-12:
python --version
pip --version

What is output of these commands?

cventour, 2018-03-12:
Python 2.7.9

pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)

uname -a

Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l
GNU/Linux

thijstriemstra, 2018-03-12:
@cventour try:

```
sudo -H pip install --upgrade --ignored-installed pip setuptools
```

cventour, 2018-03-12:
Just did that 

----------
pi@raspberrypi:~ $ sudo -H pip install --upgrade  --ignore-installed pip setuptools
Downloading/unpacking pip
  Downloading pip-9.0.1-py2.py3-none-any.whl (1.3MB): 1.3MB downloaded
Downloading/unpacking setuptools
  Downloading setuptools-38.5.2-py2.py3-none-any.whl (490kB): 490kB downloaded
Installing collected packages: pip, setuptools
Successfully installed pip setuptools
Cleaning up...
--------

btw, the error appears when I try to install from the local close. If I try with pip , I get the following message.

pi@raspberrypi:~/luma.led_matrix $ sudo pip install luma.led_matrix
Downloading/unpacking luma.led-matrix
  Downloading luma.led_matrix-1.0.8-py2.py3-none-any.whl
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/share/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/share/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/share/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 2630, in _compute_dependencies
    common = frozenset(reqs_for_extra(None))
  File "/usr/share/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 2627, in reqs_for_extra
    if req.marker_fn(override={'extra':extra}):
  File "/usr/share/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/_markerlib/markers.py", line 113, in marker_fn
    return eval(compiled_marker, environment)
  File "<environment marker>", line 1, in <module>
NameError: name 'platform_system' is not defined

thijstriemstra, 2018-03-12:
@cventour the error references `/usr/share/python-wheels/setuptools-5.5.1-py2.py3` and you really need to use setuptools 31 or newer.

cventour, 2018-03-12:
it seems that setuptools 38 is installed though

Downloading/unpacking setuptools
  Downloading setuptools-38.5.2-py2.py3-none-any.whl (490kB): 490kB downloaded
Installing collected packages: pip, setuptools
Successfully installed pip setuptools
Cleaning up...

cventour, 2018-03-18:
Hello all, 

Just for everyone to know, I cloned setuptools from its github repo and installed manually. After that I did pip list and I could see the latest version of setuptools installed.

qaisjp, 2018-07-01:
`pip install pip --upgrade` fixed this for me

mmrblack, 2019-01-05:
> @cventour try:
> 
> ```
> sudo -H pip install --upgrade --ignore-installed pip setuptools
> ```

PR #47 (2017-04-12): Display useful message when setuptools is outdated
fixes #45
commit e354fafa (2017-04-12): Display useful message when setuptools is outdated (#47)

* Display useful message when setuptools is outdated (fixes #45)
* catch ImportErrors for 3rd-party libraries in examples