rm-hull/luma.examples #46: error: argument --framebuffer: invalid choice: 'conf/st7735.conf'

thijstriemstra, 2017-04-11:
```
$ python examples/3d_box.py --f conf/st7735.conf 
usage: 3d_box.py [-h] [--config CONFIG] [--display] [--width WIDTH]
                 [--height HEIGHT] [--rotate] [--interface]
                 [--i2c-port I2C_PORT] [--i2c-address I2C_ADDRESS]
                 [--spi-port SPI_PORT] [--spi-device SPI_DEVICE]
                 [--spi-bus-speed SPI_BUS_SPEED]
                 [--gpio-data-command GPIO_DATA_COMMAND]
                 [--gpio-reset GPIO_RESET] [--gpio-backlight GPIO_BACKLIGHT]
                 [--block-orientation] [--mode] [--framebuffer] [--bgr]
                 [--h-offset H_OFFSET] [--v-offset V_OFFSET]
                 [--backlight-active]
3d_box.py: error: argument --framebuffer: invalid choice: 'conf/st7735.conf' (choose from 'diff_to_previous', 'full_frame')
```

rm-hull, 2017-04-11:
Try `-f conf/st7735.conf` rather than `--f conf/st7735.conf`:

thijstriemstra, 2017-04-11:
nice that works! it's not 100% smooth though, the light's flickering. I guess the speed needs to be increased somewhere.

rm-hull, 2017-04-12:
If the backlight is flickering check the connections are sound first.

thijstriemstra, 2017-04-12:
Hmm I spent a long time creating dupont cable header crap for that device so it'll suck if that's the issue, I'll try with some other cables and see if that fixes the flickering (it's backlight indeed, I also haven't tested with 5V yet, it's on v3.3 now).

thijstriemstra, 2017-04-12:
ps. is that confusing error when using --f something that should/can be fixed?

rm-hull, 2017-04-12:
I think that it is argparser being a bit too clever, there is a `-f` already to load the config file, and `--f` is obviously matching the framebuffer. If there isn't a strict mode... there should be.

thijstriemstra, 2017-04-12:
And it's also running on a pizero, I was thinking this could be an issue, e.g. not fast enough for a relatively large screen?

rm-hull, 2017-04-12:
its worth benchmarking the devices on different Pi's with the perfloop.py.

Note that for testing the st7735 you will _need_ to add the `--framebuffer=full-frame` flag to get realistic values:

    $  python examples/perfloop.py --f conf/st7735.conf --framebuffer=full-frame

Otherwise it will report insanely high FPS, even on a RPi0

thijstriemstra, 2017-04-12:
from `/boot/config.txt`:
```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on,i2c_baudrate=3200000
#dtparam=i2s=on
dtparam=spi=on
```

Not too great:

```
$ python examples/perfloop.py -f conf/st7735.conf --framebuffer=full_frame
Testing dislay rendering performance
Press Ctrl-C to abort test

Display: st7735
Interface: spi
Dimensions: 160 x 128
----------------------------------------
#### iter =    248: render time = 193.34 ms, frame rate = 5.17 FPS
```

thijstriemstra, 2017-04-12:
Well it got fixed by hooking it up to 5V. Thanks for the guidance @rm-hull.

closed 2017-04-12 by thijstriemstra

rm-hull, 2017-04-13:
Its interpreted python, on a slow single core SoC, and I think 160x128 is pretty much the limit of what that will render adequately. For better performance, you could look at enabling the linux framebuffer instead.

For comparison, my RPi B2 gives:

    #### iter =    341: render time = 87.32 ms, frame rate = 11.44 FPS

Note that this is the worst-case performance: it is rendering the full frame every time. Under normal usage it will work out what has changed from the last time the frame was drawn, and will only update that bit, so other demos should be reasonably snappy.