rm-hull/luma.examples #54: New user with SH1106 and "mock" error

MaryAlphaXray, 2017-05-05:
I followed the install instructions without any problems or errors. When I run "python test_sh1106.py" in the "luma.oled/tests" directory I get the following error.

pi@ChangeMe:~/luma.oled/tests $ python test_sh1106.py
Traceback (most recent call last):
  File "test_sh1106.py", line 10, in <module>
    from helpers import (serial, call, Mock, setup_function,  # noqa: F401
  File "/home/pi/luma.oled/tests/helpers.py", line 8, in <module>
    from mock import call, Mock  # noqa: F401
ImportError: No module named mock

Where did I go wrong?
Max

P.S. I'm new to programming and Git so if this is in the wrong place please let me know.

rm-hull, 2017-05-05:
Hi Max, raising an issue is absolutely the right thing to do :)

Any python scripts in the tests subfolder are unit tests that check specific bits of code, and are not really suitable for running directly - they are usually run via [tox](https://pypi.python.org/pypi/tox).

If you have a SH1106 OLED, best thing to do is double check you got the wiring connected successfully, and then if you clone the **luma.examples** repo, follow the installation instructions in the README.

Then, if it is an I2C interface on your SSH1106, first try:

    $ python examples/demo.py --display=sh1106

MaryAlphaXray, 2017-05-05:
I must say all this is very confusing. Personally, moving all the examples and not having them part of the git clone process is a big mistake.  The only way I have found to get the examples is to copy and paste each one. Is there a better way? You even told me to run "python examples/demo.py --display=sh1106" but that doesn't exist. The only thing there is README.1st which directs me back to the examples on git. And all of these programs depends on other files/programs to be in place. At this point I'm getting.....

python demo.py --display=sh1106
Traceback (most recent call last):
  File "demo.py", line 10, in <module>
    from demo_opts import device
  File "/home/pi/luma.oled/examples/demo_opts.py", line 7, in <module>
    import luma.core.emulator
ImportError: No module named emulator

I can't seem to find "luma.core.emulator" and have run out of places to look. A git clone should downloaded every needed to run the examples. 

When is the last time that you took a fresh copy of Jessie and followed your own instructions? Better yet, have someone do this that is new to Linux and the Pi.

While I'm new to Linux and the Pi, I used to do assembly language programming on computers made before most here were even born.

Thanks for all your hard work.
Max

thijstriemstra, 2017-05-06:
> The only way I have found to get the examples is to copy and paste each one. How do you clone the examples?
Please read the README. It clearly states how to clone the examples: https://github.com/rm-hull/luma.examples/#installation-instructions

> When is the last time that you took a fresh copy of Jessie and followed your own instructions? 
Once a week?

MaryAlphaXray, 2017-05-09:
I made the mistake of following the instructions found at https://luma-oled.readthedocs.io/ and didn't notice the readme in the examples. All is working ok now. 

Hard to believe that the program runner.py eats up 84% of the cpu.
With nothing running  load average: 0.00, 0.04, 0.06
Runner.py running       load average: 0.84, 0.32, 0.15

MaryAlphaXray, 2017-05-09:
Just closing this comment.

closed 2017-05-09 by MaryAlphaXray