rm-hull/luma.examples #55: Simplify pygame dependency instructions

thijstriemstra, 2017-05-08:
Instead of mentioning the pygame deps in the instructions, use `build-dep` instead.

```
$ sudo apt-get build-dep pygame
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libcaca-dev libcaca0 libdirectfb-1.2-9 libdirectfb-dev libdirectfb-extra libflac-dev libfluidsynth1 libmad0 libmad0-dev libmikmod-config libmikmod-dev libmikmod3 libogg-dev libopenal-data libopenal1 libportmidi-dev
  libportmidi0 libpython-all-dev libpython-dev libpython2.7-dev libsdl-image1.2 libsdl-image1.2-dev libsdl-mixer1.2 libsdl-mixer1.2-dev libsdl-ttf2.0-0 libsdl-ttf2.0-dev libsdl1.2-dev libsdl1.2debian libslang2-dev
  libsmpeg-dev libsmpeg0 libvorbis-dev libvorbisfile3 python-all python-all-dev python-dev python2.7-dev sharutils
0 upgraded, 38 newly installed, 0 to remove and 0 not upgraded.
Need to get 24.6 MB of archives.
After this operation, 45.5 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

rm-hull, 2017-05-08:
What version of pygame gets installed through apt?
Does it get installed for both python2 and 3?

thijstriemstra, 2017-05-08:
@rm-hull it does not install any system pygame. `build-dep` only installs the dependencies.

rm-hull, 2017-05-08:
Ah right, gotcha ... Yes that makes things much easier then :+1:

thijstriemstra, 2017-05-08:
@rm-hull can you give the command a try one of your rpi's (i dont have any stock raspbians around atm).

thijstriemstra, 2017-05-14:
Looks like it doesn't work without adding a deb-src line in /etc/apt/sources.list, meh.

```
$ sudo apt-get build-dep pygame
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: You must put some 'source' URIs in your sources.list
```

closed 2017-07-03 by thijstriemstra