rm-hull/luma.examples #57: Tox

thijstriemstra, 2017-07-03:
Tests die with this error on rpi2:

```
tests/test_demo_opts.py::test_get_device_led_matrix_all ERROR: InvocationError: '/home/rpitest/pi-test/luma.examples/.tox/py27/bin/coverage run -m py.test -vv -r w'
```

'full' run (ignore missing py3.5 error):

```
py34 runtests: commands[1] | coverage run -m py.test -vv -r w
======================================== test session starts ========================================
platform linux -- Python 3.4.2, pytest-3.1.2, py-1.4.34, pluggy-0.4.0 -- /home/rpitest/pi-test/luma.examples/.tox/py34/bin/python3.4
cachedir: .cache
rootdir: /home/rpitest/pi-test/luma.examples, inifile:
plugins: cov-2.5.1
collected 9 items 

tests/test_demo_opts.py::test_display_settings PASSED
tests/test_demo_opts.py::test_get_device_config_file_missing PASSED
tests/test_demo_opts.py::test_get_device_config_file_success PASSED
tests/test_demo_opts.py::test_get_device_unknown PASSED
tests/test_demo_opts.py::test_get_device_emulator_all PASSED
tests/test_demo_opts.py::test_get_device_led_matrix_all ERROR: InvocationError: '/home/rpitest/pi-test/luma.examples/.tox/py34/bin/coverage run -m py.test -vv -r w'
py35 create: /home/rpitest/pi-test/luma.examples/.tox/py35
ERROR: InterpreterNotFound: python3.5
```

thijstriemstra, 2017-09-15:
A new tests run on a rpi 3 (py2.7) using tox:

```
py27 runtests: commands[1] | coverage run -m py.test -vv -r wsx
=============================================================================== test session starts ================================================================================
platform linux2 -- Python 2.7.9, pytest-3.2.2, py-1.4.34, pluggy-0.4.0 -- /home/foo/luma/luma.examples/.tox/py27/bin/python2.7
cachedir: .cache
rootdir: /home/foo/luma/luma.examples, inifile:
plugins: cov-2.5.1
collected 9 items                                                                                                                                                                   

tests/test_demo_opts.py::test_display_settings PASSED
tests/test_demo_opts.py::test_get_device_config_file_missing PASSED
tests/test_demo_opts.py::test_get_device_config_file_success PASSED
tests/test_demo_opts.py::test_get_device_unknown PASSED
tests/test_demo_opts.py::test_get_device_emulator_all PASSED
tests/test_demo_opts.py::test_get_device_led_matrix_all SKIPPED
tests/test_demo_opts.py::test_get_device_lcd_all FAILED
tests/test_demo_opts.py::test_get_device_oled_all FAILED
tests/test_demo_opts.py::test_get_device_oled_interface_unknown PASSED
============================================================================= short test summary info ==============================================================================
SKIP [1] /home/foo/luma/luma.examples/tests/test_demo_opts.py:113: No module named _rpi_ws281x

===================================================================================== FAILURES =====================================================================================
_____________________________________________________________________________ test_get_device_lcd_all ______________________________________________________________________________

capsys = <_pytest.capture.CaptureFixture instance at 0x72b41a30>

    def test_get_device_lcd_all(capsys):
        """
        Load supported lcd devices one by one.
        """
        for display in cmdline.get_display_types().get('lcd'):
            try:
>               get_device(['--display', display])

tests/test_demo_opts.py:123: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
examples/demo_opts.py:51: in get_device
    device = cmdline.create_device(args)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

args = Namespace(backlight_active='low', bgr=False, block_orientation=0, config=None,...=8000000, spi_device=0, spi_port=0, transform='scale2x', v_offset=0, width=128)
display_types = OrderedDict([('oled', ['ssd1306', 'ssd1322', 'ssd1325', 'ssd1331', 'sh1106']),...ment', 'apa102']), ('emulator', ['capture', 'gifanim', 'pygame', 'asciiart'])])

    def create_device(args, display_types=None):
        """
        Create and return device.
    
        :type args: object
        :type display_types: dict
        """
        device = None
        if display_types is None:
            display_types = get_display_types()
    
        if args.display in display_types.get('oled'):
            import luma.oled.device
            Device = getattr(luma.oled.device, args.display)
            Serial = getattr(make_serial(args), args.interface)
            device = Device(Serial(), **vars(args))
    
        elif args.display in display_types.get('lcd'):
            import luma.lcd.device
            import luma.lcd.aux
            Device = getattr(luma.lcd.device, args.display)
            spi = make_serial(args).spi()
>           device = Device(spi, **vars(args))
E           TypeError: __init__() got multiple values for keyword argument 'gpio'

[... 18 lines cut ...]
    device = Device(Serial(), **vars(args))
.tox/py27/local/lib/python2.7/site-packages/luma/oled/device.py:467: in __init__
    0xA4)               # Normal dislay
.tox/py27/local/lib/python2.7/site-packages/luma/core/device.py:39: in command
    self._serial_interface.command(*cmd)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <luma.core.interface.serial.i2c object at 0x6d04fa50>, cmd = (174, 179, 242, 168, 63, 162, ...)

    def command(self, *cmd):
        """
            Sends a command or sequence of commands through to the I²C address
            - maximum allowed is 32 bytes in one go.
    
            :param cmd: a spread of commands
            :type cmd: int
            :raises luma.core.error.DeviceNotFoundError: I2C device could not be found.
            """
>       assert(len(cmd) <= 32)
E       AssertionError

.tox/py27/local/lib/python2.7/site-packages/luma/core/interface/serial.py:86: AssertionError
```

**Note** This is using the luma.core https://github.com/rm-hull/luma.core/blob/rpi-tests-90/ branch btw.

thijstriemstra, 2017-09-15:
And on Python 3.6 /RPi3 it just dies.. (again):

```==== test session starts ====
platform linux -- Python 3.6.2, pytest-3.2.2, py-1.4.34, pluggy-0.4.0 -- /home/foo/luma/luma.examples/.tox/py36/bin/python3.6
cachedir: .cache
rootdir: /home/foo/luma/luma.examples, inifile:
plugins: cov-2.5.1
collected 9 items                                                                                                                                                                   

tests/test_demo_opts.py::test_display_settings PASSED
tests/test_demo_opts.py::test_get_device_config_file_missing PASSED
tests/test_demo_opts.py::test_get_device_config_file_success PASSED
tests/test_demo_opts.py::test_get_device_unknown PASSED
tests/test_demo_opts.py::test_get_device_emulator_all PASSED
tests/test_demo_opts.py::test_get_device_led_matrix_all ERROR: InvocationError: '/home/foo/luma/luma.examples/.tox/py36/bin/coverage run -m py.test -vv -r wsx'
```

rm-hull, 2017-09-16:
Is this going to be the same wheel/setup problem again? https://github.com/rm-hull/luma.led_matrix/blob/master/setup.py#L29

thijstriemstra, 2017-10-10:
> Is this going to be the same wheel/setup problem again?

Probably.. Opened https://github.com/rm-hull/luma.led_matrix/issues/125

yoyojacky, 2023-05-12:
> _get_de

thanks for your effort on this project, it helps me a lot, I want to say thank you very much, the luma.examples helps me build my own display showing panel. that's great!

closed 2023-09-25 by thijstriemstra