rm-hull/luma.examples #59: Tests failing on travis

thijstriemstra, 2017-09-13:
https://travis-ci.org/rm-hull/luma.examples/jobs/274813513

Will take a look.

thijstriemstra, 2017-09-13:
It's very weird that RPi.GPIO and spidev are not installed on Travis. Could it be related to change in core wrt change in it's setup.py to exclude these dependencies depending on platform?

```
py35 installed: argcomplete==1.9.2,coverage==4.4.1,luma.core==1.0.1,luma.emulator==0.2.5,luma.examples==0.0.0,luma.lcd==1.0.0,luma.led-matrix==1.0.2,luma.oled==2.2.10,olefile==0.44,Pillow==4.2.1,py==1.4.34,pygame==1.9.3,pytest==3.2.2,pytest-cov==2.5.1,smbus2==0.2.0
```

rm-hull, 2017-09-14:
Yes, it must be something to do with this commit: https://github.com/rm-hull/luma.core/commit/d22184759c37dba21415a96af41deaaa6b806fca#diff-2eeaed663bd0d25b7e608891384b7298R35

Are the semantics of extend exactly the same as appending?

thijstriemstra, 2017-09-14:
```python
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> foo = [1, 2, 3]
>>> foo.extend([4, 5, 6])
>>> foo
[1, 2, 3, 4, 5, 6]
```

thijstriemstra, 2017-09-14:
Well it's very likely that's the issue.. but that means luma.core should have a new ticket.

closed 2017-09-14 by thijstriemstra

rm-hull, 2017-09-14:
Just had a thought ... for a wheel, at what point is the `setup.py` run
- when the wheel is built (on my mac), or
- when installed on the target machine ?

If the former, `platform.system()` will be Darwin so that might explain it.
Looking at the METADATA file inside the .whl:

```
Metadata-Version: 2.0
Name: luma.core
Version: 1.0.1
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Download-URL: https://github.com/rm-hull/luma.core/tarball/1.0.1
Description-Content-Type: UNKNOWN
Keywords: raspberry orange banana pi rpi opi sbc oled lcd led display screen spi i2c
Platform: UNKNOWN
Classifier: License :: OSI Approved :: MIT License
Classifier: Development Status :: 5 - Production/Stable
Classifier: Intended Audience :: Education
Classifier: Intended Audience :: Developers
Classifier: Topic :: Education
Classifier: Topic :: System :: Hardware
Classifier: Topic :: System :: Hardware :: Hardware Drivers
Classifier: Programming Language :: Python :: 2
Classifier: Programming Language :: Python :: 2.7
Classifier: Programming Language :: Python :: 3
Classifier: Programming Language :: Python :: 3.4
Classifier: Programming Language :: Python :: 3.5
Classifier: Programming Language :: Python :: 3.6
Requires-Dist: pillow (>=4.0.0)
Requires-Dist: smbus2
Requires-Dist: monotonic; python_version < "3.3"
Provides-Extra: docs
Requires-Dist: sphinx (>=1.5.1); extra == 'docs'
Provides-Extra: qa
Requires-Dist: flake8; extra == 'qa'
Requires-Dist: rstcheck; extra == 'qa'
Provides-Extra: test
Requires-Dist: pytest (>=3.1); extra == 'test'
Requires-Dist: pytest-cov; extra == 'test'
Provides-Extra: test
Requires-Dist: mock; python_version < "3.3" and extra == 'test'

...
```

thijstriemstra, 2017-09-14:
this looks like a neat solution: https://stackoverflow.com/a/42946495

```
setup(
    name="Project",
    ...
    install_requires=[
        'enum34;python_version<"3.4"',
        'pywin32 >= 1.0;platform_system=="Windows"'
    ]
)
```

rm-hull, 2017-09-14:
yup, definitely

PR #60 (2017-09-14): Fix travis build
fixes #59