rm-hull/luma.examples #71: sevensegment_demo.py fails

CoWinkKeyDinkInc, 2017-12-18:
```Version: luma.oled 2.3.1 (luma.core 1.2.1)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
------------------------------------------------------------
Traceback (most recent call last):
  File "sevensegment_demo.py", line 115, in <module>
    main()
  File "sevensegment_demo.py", line 71, in main
    seg = sevensegment(device)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 482, in __init__
    self.segment_mapper = segment_mapper or device.segment_mapper
AttributeError: 'ssd1306' object has no attribute 'segment_mapper'
```

rm-hull, 2017-12-18:
This is to be expected; ssd1306 is not a device that has the capability for supporting 7-segment displays - only max7219 or ht1621 does.

thijstriemstra, 2017-12-18:
we could put a check in there that produces a good error msg..

CoWinkKeyDinkInc, 2017-12-18:
Oh sorry about that, but yeah thanks for adding a more clear error message.

thijstriemstra, 2017-12-27:
> the capability for supporting 7-segment displays - only max7219 or ht1621 does.

@rm-hull are you sure these are the only ones? I opened for #76 this. Unfortunately we'll have to update the demo every time another supported device is added..

PR #76 (2017-12-30): show better error message for sevensegment demo
when using unsupported devices.

fixes #71
closed 2017-12-30 by thijstriemstra