rm-hull/luma.examples #72: ht1621 conf missing

thijstriemstra, 2017-12-20:
and error when trying the demo:

```
$ python examples/sevensegment_demo.py -d ht1621
/home/rpitest/pi-test/luma.core/luma/core/interface/serial.py:164: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
Traceback (most recent call last):
  File "luma.examples/examples/sevensegment_demo.py", line 115, in <module>
    main()
  File "luma.examples/examples/sevensegment_demo.py", line 70, in main
    device = get_device()
  File "/home/rpitest/pi-test/luma.examples/examples/demo_opts.py", line 61, in get_device
    device = cmdline.create_device(args)
  File "/home/rpitest/pi-test/luma.core/luma/core/cmdline.py", line 176, in create_device
    device = Device(spi, **vars(args))
TypeError: __init__() got multiple values for argument 'gpio'
```

rm-hull, 2017-12-21:
Just refreshing myself with that driver code and noticed your comment - https://github.com/rm-hull/luma.lcd/pull/45#issuecomment-296324693

Presume you're on python3?

thijstriemstra, 2017-12-22:
Yep.

rm-hull, 2017-12-22:
So it looks like the ht1621 doesn't quite conform to the same interface as the other drivers, and because `gpio` is marked as a positional argument, but it looks like the cmdline parser passes it as a keyword argument, hence the error you reported.

Presumably if you write a simple test program as described in https://luma-lcd.readthedocs.io/en/latest/python-usage.html#seven-segment-drivers, the driver works?

I think one possible 'fix' for this would be to change the constructor to be:

```python
    def __init__(self, serial_interface=None, width=6, rotate=0, WR=11, DAT=10, CS=8, **kwargs):
        super(ht1621, self).__init__(luma.lcd.const.ht1621, noop())
        self.capabilities(width, 8, rotate)
        self.segment_mapper = dot_muncher
        self._gpio = kwargs.get('gpio') or self.__rpi_gpio__()
```

This _should_ be compatible with the cmdline/demo-opts and continue to work for direct invocation, but I'm not in a position to test this at the moment, could you try it?

rm-hull, 2017-12-22:
As to your other comment, in the headline about a missing config -- as everything is defaulted in the constructor, the only thing a config file would contain is:

```
-d ht1621
```

We can add it if you like?

thijstriemstra, 2017-12-23:
> Presumably if you write a simple test program as described in https://luma-lcd.readthedocs.io/en/latest/python-usage.html#seven-segment-drivers, the driver works?

Yep!

> but I'm not in a position to test this at the moment, could you try it?

Same here, but will do.

> We can add it if you like?

Sound good (for newbies).

thijstriemstra, 2017-12-27:
Created #75 for the config file, still need to test the suggested fix for that error.

thijstriemstra, 2017-12-27:
Closed too soon, that PR shouldn't have closed this.

thijstriemstra, 2017-12-30:
@rm-hull tried your fix and getting the same error:

```
$ python examples/sevensegment_demo.py -d ht1621
Version: luma.lcd 1.0.3 (luma.core 1.2.1)
Display: ht1621
Interface: i2c
Dimensions: 128 x 64
------------------------------------------------------------
/home/rpitest/pi-test/luma.core/luma/core/interface/serial.py:164: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
Traceback (most recent call last):
  File "examples/sevensegment_demo.py", line 121, in <module>
    main()
  File "examples/sevensegment_demo.py", line 71, in main
    device = get_device()
  File "/home/rpitest/pi-test/luma.examples/examples/demo_opts.py", line 63, in get_device
    device = cmdline.create_device(args)
  File "/home/rpitest/pi-test/luma.core/luma/core/cmdline.py", line 176, in create_device
    device = Device(spi, **vars(args))
TypeError: __init__() got multiple values for argument 'gpio'
```

the traceback also doesn't refer to luma.lcd specifically..

rm-hull, 2017-12-30:
Also the constructor in the suggested fix different:

```diff
-def __init__(self, gpio=None, width=6, rotate=0, WR=11, DAT=10, CS=8, **kwargs):
+def __init__(self, serial_interface=None, width=6, rotate=0, WR=11, DAT=10, CS=8, **kwargs):
```

thijstriemstra, 2017-12-30:
ah! let me check.

rm-hull, 2018-01-27:
If the demo is printing weird characters then that implies the segments on your board are wired up differently, and that we need a different segment mapper that that provided: https://github.com/rm-hull/luma.lcd/blob/master/luma/lcd/segment_mapper.py

I remember having to implement a translation layer for the [NeoSegments](https://neosegment.com/) (see https://github.com/rm-hull/luma.led_matrix/blob/master/luma/led_matrix/device.py#L512-L538) - maybe we need something similar here?

thijstriemstra, 2018-01-27:
@rm-hull you have a copy of this device as well right? Could you give it a try at some point?

thijstriemstra, 2018-01-27:
ps. using suggested code (and not the examples) worked fine for me (only tested numerals though).

commit ab4a4429 (2017-12-27): add ht1621 config file (#75)

Fixes #72