rm-hull/luma.examples #73: SSD1331 hangs

sgjava, 2017-12-24:
Using ODROID C1+ and the following works fine:

```
import time
from RPi import GPIO
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1331

GPIO.setmode(GPIO.BOARD)
GPIO.setwarnings(False)
serial = spi(port=0, device=0, gpio=GPIO, gpio_DC=18, gpio_RST=22)
device = ssd1331(serial)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white")
    draw.text((3, 3), "Hello", fill="white")

time.sleep(10)
```

Running any of the examples with the following hangs the program:

```
sudo python examples/invaders.py -d ssd1331 -i spi --spi-port 0 --spi-device 0 --gpio-data-command 18 --gpio-reset 22 --width 96 --height 64
Version: luma.oled 2.3.1 (luma.core 1.2.1)
Display: ssd1331
Interface: spi
Dimensions: 96 x 64
------------------------------------------------------------
```

rm-hull, 2017-12-24:
If you run the perfloop.py example, do you see the FPS being reported on the console?

rm-hull, 2017-12-24:
Ok, so the reason the examples dont work for you is that when it initializes, it does so with BCM numbering rather than BOARD (like your working example does). See https://github.com/rm-hull/luma.core/blob/master/luma/core/lib.py#L24.

If odroid supports BCM pin numbers, try substituting the right values on the command line and it _should_ work as expected.

sgjava, 2017-12-24:
OK, cool, I'l try BCM after looking at https://github.com/swkim01/RPi.GPIO-OdroidC1 and I'll reply back here...

sgjava, 2017-12-24:
Haha, you rock, Merry Xmas!

sudo python demo.py -d ssd1331 -i spi --spi-port 0 --spi-device 0 --gpio-data-command 24 --gpio-reset 25 --width 96 --height 64

closed 2017-12-24 by sgjava