rm-hull/luma.examples #77: I2C device not found on address: 0x3C

mikolajzieba, 2018-01-05:
I followed your instructions and getting error: 
```I2C device not found on address: 0x3C```

Result of the `i2cdetect -y 1`
```     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                     ```

Could you give me a tip where to look for a source of the problem?

rm-hull, 2018-01-05:
Can you show the command you were trying to run when you got the error?

mikolajzieba, 2018-01-08:
I'm sorry, It realized that it was a problem with my GND line, so Issue can be closed.

January 5, 2018 3:02 PM, "Richard Hull"  wrote:
	Can you show the command you were trying to run when you got the error?