rm-hull/luma.examples #78: running second OLED screen on SPI1 example

overthrowrobotics, 2018-01-07:
I have two SSD1351 screens. I've tested them with code from Adafruit and they both are working. The first screen is connected to SPI0 and is working fine with luma examples. 

The second screen is connected to SPI1 on GPIO20 (MOSI), GPIO21 (SCLK), GPIO16 (CE2) and is sharing DC and Reset with the first screen on pins 5 and 6.  

In the documentation there's an option called --gpio GPIO. I'm not sure how to use it. I've tried putting in --gpio gpio20 and --gpio 20 and neither works. What would be the correct settings?

rm-hull, 2018-01-07:
Are you trying to get output on both of the screens at the same time? If so, then the examples are not designed for that and it probably wont work without some extra effort.

If you just want to make sure the second screen is working on SPI1, then you should be able to call something like:

```
python examples/demo.py --f conf/ssd1351.conf --spi-device=1 --spi-port=1
```

That, _I think_, will make it use `/dev/spi1.1`, but I can never remember which way round the SPI device and port entries are.

The `--gpio` flag is to swap out the RPi.GPIO library for something else (e.g. OPi.GPIO) for those devices where RPi.GPIO wont work.

If you enter `python examples/demo.py -h`, you should get a help page with brief instructions on the available flags.

overthrowrobotics, 2018-01-07:
Running : `python demo.py --f conf/ssd1351.conf --spi-device=1 --spi-port=1 --gpio-data-command 5 --gpio-reset 6`

Gives this: 
`demo.py: error: argument --framebuffer: invalid choice: 'conf/ssd1351.conf' (choose from 'diff_to_previous', 'full_frame')`

This runs without error but I don't get anything on the screen. I also tried --spi-device=0 while leaving the spi-port=1 and that didn't work either. 
`python demo.py --display ssd1351 --spi-device=1 --spi-port=1 --gpio-data-command 5 --gpio-reset 6 --interface spi --width 128 --height 128`

rm-hull, 2018-01-08:
Sorry, my bad: that command should've had `-f` not `--f`.

> The second screen is connected to SPI1 on GPIO20 (MOSI), GPIO21 (SCLK), GPIO16 (CE2) and is sharing DC and Reset with the first screen on pins 5 and 6.

Did you specifically enable SPI1 ? see: https://tutorials.technology/tutorials/69-Enable-additonal-spi-ports-on-the-raspberrypi.html -- what does `ls -l /dev/spi*` show?

Alternatively, you should be able daisychain the 2nd screen onto the SPI0 clock and data pins, but the hook up the device's chip select to CE1.

overthrowrobotics, 2018-01-08:
Both screens are working with the adafruit library so SPI1 must be turned on. 

`crw-rw---- 1 root spi 153, 0 Jan  7 07:11 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Jan  7 07:11 /dev/spidev0.1
crw-rw---- 1 root spi 153, 4 Jan  7 07:11 /dev/spidev1.0
crw-rw---- 1 root spi 153, 3 Jan  7 07:11 /dev/spidev1.1
crw-rw---- 1 root spi 153, 2 Jan  7 07:11 /dev/spidev1.2
`

python demo.py -f conf/ssd1351.conf --spi-device=0 --spi-port=1 --gpio-data-command 5 --gpio-reset 6 --interface spi --width 128 --height 128
gives me: 
`IOError: [Errno 2] No such file or directory: 'conf/ssd1351.conf'
`

overthrowrobotics, 2018-01-08:
Wait. I noticed spidev1.2 so I just tried --spi-port=1 --spi-device=2 and the second screen turned on. 

I tried running the script twice using & to run in background. Half of the output on one screen showed up on the other screen and it was shifted down. 

Would it be a ton of work to get both screens working simultaneously?

rm-hull, 2018-01-25:
You should be able to get two screens running simultaneously. I have done it with ssd1306's and an i2c multiplexer, but same principle applies to spi.

You can't expect to have two scripts both accessing spi without some coordination, as like you discovered, images will get corrupted between the two. You should be able to create a python script which creates two displays each with its own spi config, and then update each screen in turn.

See also https://github.com/rm-hull/luma.core/issues/55

elpimous, 2018-01-25:
Hi all.
Yes I confirm..
My both ssd1351work in parallel (Y cables)
No latency.
Have a nice day.

thijstriemstra, 2018-02-07:
@overthrowrobotics any news?

overthrowrobotics, 2018-02-08:
I gave up on it for now and started playing with the library that Adafruit has for their raspberry pi board that connects two SSD1351.

thijstriemstra, 2018-02-13:
If you find any issues, open a new ticket in the luma.oled issue tracker.

closed 2018-02-13 by thijstriemstra