rm-hull/luma.examples #80: Performance numbers for ssd1322 screen

megatron-uk, 2018-01-24:
I'm building a digital dashboard and data logger for my car and have a couple of screens that are supported by Luma.

I have 2x ssd1306 128x64 1.3" screens on the Pi 3 I2C bus. Performance when testing with the Luma examples is 25ish FPS which is fine for the purposes of using them as supplementary data gauges.

I've also added the 256x64 ssd1322 as a main display which runs great (it's a lovely screen), but the performance data seems to think it's running at over 1500fps. Has anyone else submitted feedback on this screen? It seems quick, but I don't quite believe it's that quick!

rm-hull, 2018-01-25:
No it will be slower than that. If you are testing with the perfloop script, rerun with the `--framebuffer=full_frame` flag. By default, the ssd1322 driver only redraws difference, and perfloop doesn't change its image between frames hence the artificial high fps.

Would be interested to see pics of the finished digital dash when its installed in your car!

thijstriemstra, 2018-01-25:
> Would be interested to see pics of the finished digital dash when its installed in your car!

+1

megatron-uk, 2018-02-03:
Okay, retested, using the following command line:
```python3 perfloop.py -d ssd1322 --width 256 --height 64 -i spi --framebuffer=full_frame```

I get ~15fps, which is not as good as I'd hoped.

This is on Raspbian current, using a Raspberry Pi 3, with the ondemand governor. I tried increasing the SPI bus speed from 8MHz to 16MHz and it increased draw speed by a negligible ~1fps. 32MHz results in an unstable display on the OLED module. I'm using 4 wire SPI mode, seems like this is the fastest it will go.

It's a little disappointing, since the sh1106 screens I've got will push over twice that on I2C, admitedly with an increased I2C baudrate of either 1 or 1.2MHz.

A demo of my current dashboard/datalogging monitor is up here:

https://www.youtube.com/watch?v=zwGecxEOALc

https://www.youtube.com/watch?v=ULw-blmC39o

Project repo: https://github.com/megatron-uk/PyCosworth

rm-hull, 2018-02-04:
With the **full-frame** setting, remeber this is the _worst-case_ FPS, inasmuch as the whole screen is forced to be re-rendered every frame. 

With **diff-to-previous** (the default), it _could be_ as bad as the full-frame setting, but only if the whole frame changed. However, in a lot of cases it is much better, esp. if only a small part of the screen changed, and there is a lot of black background (i.e. fewer pixels to draw).

Also, consider that the SSD1322 is doing 16-bit greyscale conversion and has to process 2x as many pixels as the SH1106.

Actually, if you don't need greyscale, try setting the `mode="1"` when initializing the device in your code. It would be interesting to see what you get with:

```
python3 perfloop.py -d ssd1322 --width 256 --height 64 -i spi --framebuffer=full_frame --mode=1
```

Also, IIRC python2 was faster than python3.

Good videos BTW - chapeau!

thijstriemstra, 2018-02-04:
> Also, IIRC python2 was faster than python3.

:(

megatron-uk, 2018-02-04:
Yep, that's faster - I'm hitting 29-30fps with --mode=1. As most of my use case is line drawing and text, I'll probably not benefit from the greyscale display mode, but the speed boost probably _would_ be useful to me :)

My initialisation now looks like:
serial = spi(port = 0, device = 0, bus_speed_hz=16000000)
luma_driver = ssd1322(serial_interface = serial, mode = "1", framebuffer = "diff_to_previous")

closed 2018-02-04 by megatron-uk