rm-hull/luma.examples #85: ValueError: The channel sent is invalid

attilagyorffy, 2018-02-17:
Hi,

I'm trying to run the example project using a SSD1322 display on a FriendlyCore OS on a NanoPi NEO Air.

Running `sudo python examples/sys_info.py -f conf/myconfig.cfg` I'm getting an error:

```
Version: luma.oled 2.3.1 (luma.core 1.5.0)
Display: ssd1322
Interface: spi
Dimensions: 256 x 64
------------------------------------------------------------
Traceback (most recent call last):
  File "examples/sys_info.py", line 111, in <module>
    device = get_device()
  File "/home/pi/luma.examples/examples/demo_opts.py", line 63, in get_device
    device = cmdline.create_device(args)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/cmdline.py", line 178, in create_device
    device = Device(Serial(), **vars(args))
  File "/usr/local/lib/python2.7/dist-packages/luma/core/cmdline.py", line 160, in spi
    gpio=self.gpio or GPIO)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 255, in __init__
    bitbang.__init__(self, gpio, transfer_size, DC=gpio_DC, RST=gpio_RST)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 153, in __init__
    self._DC = self._configure(kwargs.get("DC"))
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 164, in _configure
    self._gpio.setup(pin, self._gpio.OUT)
ValueError: The channel sent is invalid on a NanoPi
```

I have the following config defined:

```
--display=ssd1322
--interface=spi
--width=256
--height=64
--interface=spi
--spi-bus-speed=16000000
--spi-device=0
--gpio-data-command=201
--gpio-reset=1
```

I believe the FriendlyARM OS comes with the RPI.GPIO python package preinstalled which *should* have the right physical/gpio pin mappings. During the `luma.examples` pip install I was keeping an eye on the install output and have seen no output that would suggest that the RPI.GPIO module was overwritten.

I think I have the wiring set up correctly, using the [NHD-2.8-25664UCB2](http://www.newhavendisplay.com/nhd2825664ucb2-p-3539.html) display.

I wonder what the error message really means and what I could do to debug the problem forward.
Any help would be appreciated, thanks.

PS: I did try this on an Armbian OS (which doesn't have the RPi.GPIO module preinstalled, I got a similar error, only it said `ValueError: The channel sent is invalid on a Raspberry Pi" (this makes me believe that the RPI.GPIO library should be working fine?)

thijstriemstra, 2018-02-17:
interesting error message..

thijstriemstra, 2018-02-17:
@attilagyorffy can you git clone the repository for luma.core and run the unit tests there? We should start at the core and go from there..

attilagyorffy, 2018-02-18:
@thijstriemstra Here you go, this output is a bit meaningless to me but I hope you may be able to she some light. Thanks in advance.

https://gist.github.com/attilagyorffy/a74c833172b0106781cedcebced03255

rm-hull, 2018-02-18:
The error stack trace points to the RPi.GPIO library not accepting the value of 201 when trying to set up the data/command pin. What lead you to selecting that value of 201? Is there a board pin reference for nanopi that you used? Can you share a link?

If 201 is definitely correct, it could be that the examples are passing through the pin as a string when a number is required (or vise-versa)

I would start by writing a small test program just to exercise the GPIO to try and isolate the problem

attilagyorffy, 2018-02-18:
@rm-hull 201 is the right GPIO pin according to the [Nano Pi NEO Air layout diagram](http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Air#Diagram.2C_Layout_and_Dimension)

I've also tested this by manually exporting, setting the direction and the value of the pin via dev:

```
echo "201" >/sys/class/gpio/export
echo "out" >/sys/class/gpio/gpio201/direction
echo "1" > /sys/class/gpio/gpio201/value
```

The right physical pin is now turned on. (Measured the voltage with an actual device).

What I don't understand is how luma.core interacting with the pin itself. Is it sending the same data via the linux `/dev/` device?

attilagyorffy, 2018-02-23:
OK, so first of all the part that I did not get right was that I used the linux GPIO pin numbering scheme over the physical header numbers. (Which means 201 should have been 18 and 1 should have been 22. See the NanoPI Neo Air layout).

Now I have an example program that I can run without any error messages, yet the display is blank and shows no output. (I've validated the connections, these should be just fine. I can turn on the display outside of the context of the luma project):

```python
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)

from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1322
from PIL import ImageDraw

serial = spi(port=0, gpio=GPIO, gpio_DC=18, gpio_RST=22, bus_speed_hz= 16000000)
device = ssd1322(serial)
with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="white")
```

Important: I'm passing in an RPI.GPIO implementation that has been previously instructed to use the board numbering, otherwise the code fails upon setting the spi connection.

Now that I think the GPIO connection is fine, I am looking for a way to further debug the code.

From what I've seen in a previous issue, the bus speed etc should be fine unless you tell me otherwise. https://github.com/rm-hull/luma.oled/issues/90

thijstriemstra, 2018-02-23:
@attilagyorffy not sure using BOARD mode is a good idea since we hardcode it to BCM: https://github.com/rm-hull/luma.core/blob/master/luma/core/lib.py#L24 But maybe I'm misunderstand.

bishii, 2018-09-01:
FYI...
Changing to:
GPIO.setmode(GPIO.BCM)

fixed it for me. Cheers

rm-hull, 2020-10-30:
@attilagyorffy did you solve this issue? Im going to close this ticket. Please re-open if you need further assistance

closed 2020-10-30 by rm-hull

attilagyorffy, 2020-10-31:
> @attilagyorffy did you solve this issue? Im going to close this ticket. Please re-open if you need further assistance
To be honest I gave up on the project and don't remember honestly where I've left it off, sorry and thanks anyway.