rm-hull/luma.examples #87: Display image with canvas

ayhlai, 2018-05-22:
From the examples I have seen and tried , so display a picture such as png, I need to use the device.display method.  Is it possible to display an image within a canvas. 

What I would like to do is to display an image inside
` with canvas(device, dither=True) as draw:`
so that I can have a mixture of text and png in one screen, similar to colors.py example but displaying the canvas and image at the same time.

Is it possible?  

Thanks for this great library.

dummys, 2018-07-04:
same question here

thijstriemstra, 2018-07-04:
ping @rm-hull

iv-med, 2019-01-05:
I have the same question!

rm-hull, 2019-01-05:
Have a look at https://luma-core.readthedocs.io/en/latest/render.html#luma.core.render.canvas - you'll notice that it takes an optional background image as one of its arguments. Construct a pillow image from a PNG, pass that in as a background, and then draw on it.

iv-med, 2019-01-05:
> Have a look at https://luma-core.readthedocs.io/en/latest/render.html#luma.core.render.canvas - you'll notice that it takes an optional background image as one of its arguments. Construct a pillow image from a PNG, pass that in as a background, and then draw on it.

Ok, but if I want to place some images on the canvas?

rm-hull, 2020-10-30:
@iv-med if you want to just display images, there is no need to use the canvas, just call `device.display(image)` .. the canvas is only there for drawing on.