rm-hull/luma.examples #89: ux1701x device demo - unexpected screen output

wangdino, 2018-08-22:
Hello, first of all thanks for the great lib and examples.

I have a 12864 LDC with UC1701X IC and I have it connected to a RPi3B+ (running latest Raspbian) following the wiring documentation here:
https://luma-lcd.readthedocs.io/en/latest/install.html#connecting-the-display

[image]

Also following the instructions I've installed `luma.lcd` and cloned `luma.examples` locally.

So I tried to run `demo.py` under examples folder, giving the GPIO pins based on how the wiring was connected:
`python examples/demo.py -d uc1701x -i spi --gpio-data-command 23 --gpio-reset 24`

No error was output, I think it's only showing the routine output:

> Version: luma.lcd 1.0.3 (luma.core 1.7.2)
> Display: uc1701x
> Interface: spi
> Dimensions: 128 x 64
> ------------------------------------------------------------
> Testing basic canvas graphics...
> Testing contrast (dim/bright cycles)...
> Testing display ON/OFF...
> Testing clear display...
> Testing screen updates...

However, on the other side - on the actual LCD screen, there was no response / action during most of the time, except:
When "Testing display ON/OFF..." the screen flickered like this:
[image]
When "Testing screen updates..." the following characters were shown:
[image]
And the characters moved from right side to the left side.

So I'm guessing the wiring connection is fine because otherwise nothing would be shown on screen. But since there's no error message it's difficult for me to further diagnose where could go wrong.

Any insight is appreciated.

thijstriemstra, 2018-12-30:
ping @rm-hull

rm-hull, 2020-10-30:
Looks like there is some cross-talk on the SPI lines to me. Is there another program also using the same SPI bus ?

wangdino, 2020-11-01:
It's been a while so I can't say for sure but it's highly unlikely. I was testing using a fresh system (raspbian) so there was hardly anything that would be fighting on the same SPI bus.