rm-hull/luma.examples #95: TypeError running examples for ssd1325 with Python 3

johnnyvenom, 2018-12-12:
Hi, all of the examples are failing on RPi 3 Model B (Raspbian Stretch) for ssd1325 display, when running with Python 3 (Python v3.5.3). I followed the installation instructions, and had to also install libopenjp2-7-dev. Now I get the following error when I try to run any example: 

`
pi@raspberrypi:~/luma.examples/examples $ python3 welcome.py -d ssd1325 -i spi
Version: luma.oled 3.0.0 (luma.core 1.8.2)
Display: ssd1325
Interface: spi
Dimensions: 128 x 64
------------------------------------------------------------
/usr/local/lib/python3.5/dist-packages/luma/core/interface/serial.py:164: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
Traceback (most recent call last):
  File "welcome.py", line 229, in <module>
    device = get_device()
  File "/home/pi/luma.examples/examples/demo_opts.py", line 63, in get_device
    device = cmdline.create_device(args)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/cmdline.py", line 179, in create_device
    device = Device(Serial(), **vars(args))
  File "/usr/local/lib/python3.5/dist-packages/luma/oled/device/__init__.py", line 469, in __init__
    **kwargs)
TypeError: __init__() got multiple values for keyword argument 'framebuffer'
`

I've compiled and run with Python 2.7 on a different (identical) RPi without issue. Also, I can run the simple "Hello World" script found at https://luma-oled.readthedocs.io/en/latest/python-usage.html with Python 3.5 without issue as well. 

Any advice? Thanks.

rm-hull, 2018-12-13:
I think you might've run into a regression issue as a result of https://github.com/rm-hull/luma.oled/pull/203. I've had a brief look and think I know what the cause us, but I'll investigate further, and issue a patched version shortly if necessary.

closed 2018-12-15 by rm-hull

johnnyvenom, 2018-12-18:
Thanks for the quick fix! Have yet to test it but if I have further issue will report it.