rm-hull/luma.examples #96: Command line description for --interface

thomasw-mitutoyo-ctl, 2018-12-30:
The help says that there's an optional parameter

```[--interface]```

Later it says

```
--interface , -i      Serial interface type. Allowed values are: i2c, spi,
                        bitbang (default: i2c)
```

So, if that parameter needs values, it should read

```[--interface VALUE]```

where VALUE can be replaced by i2c, spi or bitbang.

thomasw-mitutoyo-ctl, 2018-12-30:
The same applies for `--display`, `--rotate`, `--block-orientation`, `--mode`, `--framebuffer`, `--backlight-active` and `--transform`

thijstriemstra, 2018-12-30:
Looks like this is a limitation of the argparse Python module.. not sure what to do about it.

thijstriemstra, 2018-12-30:
Yea, removing metavar fixes it:

```
  --interface {i2c,spi,bitbang}, -i {i2c,spi,bitbang}
                        Serial interface type. Allowed values are: i2c, spi,
                        bitbang (default: i2c)
```

thijstriemstra, 2018-12-30:
> The help says that there's an optional parameter

it's optional because you can use the `--config` parameter and specify all params there.

closed 2019-01-01 by rm-hull