rm-hull/luma.examples #100: welcome.py not displaying any output

fmentiplay, 2019-02-07:
Hello 
Thank you for producing the Luma library. I am having a problem with the welcome.py script.

My Hardware/ Software setup is as follows........
Hardware: Raspberry Pi 3B V1.2
OS:   Raspbian GNU/Linux 9 (stretch)
Kernel: Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

I have a OLED 128x128 dislpay (with ssd1351 driver) connected as suggested at
 https://luma-oled.readthedocs.io/en/latest/hardware.html

I have cloned luma.oled, luma.docs, luma.examples

I have run most of the examples **successfully** BUT when I run welcome.py I get no display output.

This is the command I run.....

pi@raspberrypi:~/luma.examples/examples $ python3 welcome.py --display ssd1351 --interface spi --spi-port 0 --spi-device 0 --width 128 --height 128 --framebuffer=full_frame

This the output....
Version: luma.oled 3.1.0 (luma.core 1.8.3)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------

It doesn't give any errors BUT nothing is displayed on the OLED. Is it a font issue?

If I run this simple script it works fine.

> from time import sleep
> from luma.core.interface.serial import spi
> from luma.core.render import canvas
> from luma.oled.device import ssd1351
> 
> from PIL import ImageFont
> 
> serial = spi(device=0, port=0)
> device = ssd1351(serial)
> 
> font = ImageFont.truetype("/home/pi/luma.examples/examples/fonts/pixelmix.ttf",12)
> 
> with canvas(device) as draw:
>     draw.rectangle(device.bounding_box, outline="white", fill="green")
>     draw.text((20,40), "Hello World", font=font, fill="white")
>     draw.text((20,60), 'Cool Stuff', font=font, fill="white")
>     
> sleep(1000)

Any help with this issue would be greatly appreciated
Thanks in advance

thijstriemstra, 2019-12-19:
@fmentiplay can you remove the `--framebuffer=full_frame` option, it's the only difference I can see (besides the font change).

rm-hull, 2020-10-25:
Fixed with commit da48c2e

closed 2020-10-25 by rm-hull