rm-hull/luma.examples #101: ssd1327 and framebuffer='diff_to_previous' not working

kayjayme, 2019-03-07:
My Pi zero and i2c ssd1327 128x128 module shows garbage text when framebuffer='diff_to_previous' is used on the demo.py example. I'd like to use it for faster partial display updates

CosmoFox, 2019-03-23:
pi zero H5 ssd 1327 same issue!!
Workaround - use "full_frame", but it`s like slideshow!

kayjayme, 2019-03-23:
Updates are so slow that 'full_frame' is not really much use unfortunately.

rm-hull, 2019-03-24:
I'll dig out my ssd1327, wire it up and give it a go. If you could remind me though: is your display configured as SPI, and if so what is the bus speed you are using it with? Also what version of Luma.core and Luma..oled are you using?

kayjayme, 2019-03-24:
My Display is I2C running at 'baudrate=400000'. I can't access my system just now but downloaded (github) 2 weeks ago

rm-hull, 2019-03-24:
I’m going to transfer this issue to luma.oled as the bug will be in that module rather than luma.examples