rm-hull/luma.examples #102: ssd1306 Histogram example

Tkopic001, 2019-03-15:
Hello,

I made this simple system monitor screen for my OrangePi Zero and ssd1306(128x64)
It shows a CPU load histogram , RAM usage bar, CPU  thermometer bar and system uptime.

I saw a histogram in "TODO" somewhere in your examples while learning how to use the library so i tough maybe you guys can use this in this repo.

I'm pretty much a newbie when it comes to programming so i'm sure this can be optimized

[sysinfo_ssd1306_128x64.zip](https://github.com/rm-hull/luma.examples/files/2970030/sysinfo_ssd1306_128x64.zip)

Tkopic001, 2019-03-15:
Improved version with some bugs fixed:

[sysinfo_ssd1306_128x64.zip](https://github.com/rm-hull/luma.examples/files/2972355/sysinfo_ssd1306_128x64.zip)

thijstriemstra, 2019-03-15:
thanks @Tkopic001. Can you paste that in a new file (e.g. https://github.com/rm-hull/luma.examples/new/master/examples) and open a pull request? that way we can properly attribute it to you.

Tkopic001, 2019-03-15:
Hello,

Thank you for your answer, I have added the file

Kind regards,

PR #103 (2019-03-16): add histogram example
https://github.com/rm-hull/luma.examples/issues/102#issuecomment-473391918

I added a new file with my example system monitor

fixes #102
closed 2019-03-16 by rm-hull