rm-hull/luma.examples #107: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE

dezmob, 2019-07-29:
On a Pi 3B+ running buster

`pi@raspberrypi:~/luma.examples $ sudo -H pip3 install -e .`

```bash
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/luma.examples
Collecting argcomplete (from luma.examples==0.0.0)
  Downloading https://files.pythonhosted.org/packages/4d/82/f44c9661e479207348a979b1f6f063625d11dc4ca6256af053719bbb0124/argcomplete-1.10.0-py2.py3-none-any.whl
Requirement already satisfied: luma.core>=1.8.0 in /usr/local/lib/python3.7/dist-packages (from luma.examples==0.0.0) (1.12.0)
Collecting luma.emulator>=1.0.2 (from luma.examples==0.0.0)
  Downloading https://files.pythonhosted.org/packages/96/8c/3631c7906c66937d714f8490af842cf5e55fd3ff2ffa00c6f4663e60286f/luma.emulator-1.2.0-py2.py3-none-any.whl
Collecting luma.lcd>=1.0.3 (from luma.examples==0.0.0)
  Downloading https://files.pythonhosted.org/packages/21/2b/7eda9460efbf2999823bd3912aacad7988aca7055ea4bfaacc6c26b56b6b/luma.lcd-2.1.0-py2.py3-none-any.whl
Collecting luma.led_matrix>=1.0.7 (from luma.examples==0.0.0)
  Downloading https://files.pythonhosted.org/packages/76/e2/43a82bbe5a88b6a566c8a72d60ff6cc89b2d06581b640d3766498fc6cc59/luma.led_matrix-1.4.0-py2.py3-none-any.whl
Requirement already satisfied: luma.oled>=3.1.0 in /usr/local/lib/python3.7/dist-packages (from luma.examples==0.0.0) (3.3.0)
Requirement already satisfied: pyftdi; python_version >= "3.5" in /usr/local/lib/python3.7/dist-packages (from luma.core>=1.8.0->luma.examples==0.0.0) (0.29.4)
Requirement already satisfied: RPI.GPIO; platform_system == "Linux" in /usr/local/lib/python3.7/dist-packages (from luma.core>=1.8.0->luma.examples==0.0.0) (0.7.0)
Requirement already satisfied: spidev; platform_system == "Linux" in /usr/local/lib/python3.7/dist-packages (from luma.core>=1.8.0->luma.examples==0.0.0) (3.4)
Requirement already satisfied: pillow>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from luma.core>=1.8.0->luma.examples==0.0.0) (6.1.0)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.7/dist-packages (from luma.core>=1.8.0->luma.examples==0.0.0) (0.2.3)
Collecting pygame (from luma.emulator>=1.0.2->luma.examples==0.0.0)
  Downloading https://www.piwheels.org/simple/pygame/pygame-1.9.6-cp37-cp37m-linux_armv7l.whl (3.6MB)
    90% |████████████████████████████▉   | 3.3MB 131kB/s eta 0:00:03
Collecting ws2812; platform_machine == "armv7l" and platform_system == "Linux" (from luma.led_matrix>=1.0.7->luma.examples==0.0.0)
  Downloading https://www.piwheels.org/simple/ws2812/ws2812-1.0.0-cp37-cp37m-linux_armv7l.whl (95kB)
    100% |████████████████████████████████| 102kB 120kB/s
Requirement already satisfied: rpi-ws281x; platform_machine == "armv7l" and platform_system == "Linux" in /usr/local/lib/python3.7/dist-packages (from luma.led_matrix>=1.0.7->luma.examples==0.0.0) (4.2.2)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi; python_version >= "3.5"->luma.core>=1.8.0->luma.examples==0.0.0) (3.4)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi; python_version >= "3.5"->luma.core>=1.8.0->luma.examples==0.0.0) (1.0.2)
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pygame from https://www.piwheels.org/simple/pygame/pygame-1.9.6-cp37-cp37m-linux_armv7l.whl#sha256=173e72b7de36c5350af15da50fb391e03f7966bfaf804ae591481bfbb7045399 (from luma.emulator>=1.0.2->luma.examples==0.0.0):
        Expected sha256 173e72b7de36c5350af15da50fb391e03f7966bfaf804ae591481bfbb7045399
             Got        0a2cbdd4c10cb9edaf1883c5fc42eb0aa1dc7721ce0934919ac9d14d3c7522fd
```

rm-hull, 2019-07-29:
It looks as though that warning is shown because of pygame specifically not matching the expected hash, and it is possible you have a corrupted download for that wheel. You could try what is suggested here: https://stackoverflow.com/questions/40183108/python-packages-hash-not-matching-whilst-installing-using-pip

@thijstriemstra any ideas?

thijstriemstra, 2019-07-29:
yea if `pip3 install --no-cache-dir pygame` doesn't work, i would ignore this error. I think it's related to the `www.piwheels.org` repository that pip uses on raspbian..

thijstriemstra, 2019-12-19:
did you get it to work @dezmob?

thijstriemstra, 2020-01-12:
no feedback, closing.

closed 2020-01-12 by thijstriemstra