rm-hull/luma.examples #109: carousel.py shifts too fast

bk4nt, 2019-12-07:
Hello,

I've just installed luma and tested some examples on a Pi 4. carousel.py scrolls a way to fast on it, making the display unreadable, and by the way, eats up almost a full core for the scrolling job.

An option to tweak the scrolling speed and stops could be usefull? I added it that way, sure a dirty way:

```
$ diff ./carousel.py carousel.modded.py -u
--- ./carousel.py       2019-12-07 00:03:41.623544517 +0100
+++ carousel.modded.py  2019-12-07 01:15:13.804087586 +0100
@@ -16,6 +16,7 @@
 """

 import psutil
+import time

 from demo_opts import get_device
 from luma.core.virtual import viewport, snapshot
@@ -36,8 +37,10 @@
 def pause_every(interval, generator):
     try:
         while True:
+            time.sleep(0.01)
             x = next(generator)
             if x % interval == 0:
+                time.sleep(2)
                 for _ in range(20):
                     yield x
             else:

```

Best regards

thijstriemstra, 2019-12-07:
thanks @bk4nt can you make a pull request?

bk4nt, 2019-12-07:
That is a bad idea. I never made a pull request. Neither do I understand every lines of caroussel.py.

I had a better look. It seems for me that caroussel.py makes its pauses by calling yield only, plus a very basic for loop, which makes it by the way scroll and pause according to CPU speed.

As the yields and the for loop aren't needed any more, I now removed them:
```
$ diff ./carousel.py caroussel.speed.py -u
--- ./carousel.py       2019-12-07 00:03:41.623544517 +0100
+++ caroussel.speed.py  2019-12-07 16:49:31.302037659 +0100
@@ -16,6 +16,7 @@
 """

 import psutil
+import time

 from demo_opts import get_device
 from luma.core.virtual import viewport, snapshot
@@ -38,10 +39,12 @@
         while True:
             x = next(generator)
             if x % interval == 0:
-                for _ in range(20):
-                    yield x
+                # Widget pause
+                time.sleep(2.5)
             else:
-                yield x
+                # Scrolling speed
+                time.sleep(0.025)
+            yield x
     except StopIteration:
         pass

```

bk4nt, 2019-12-07:
I now noticed two issues with the code line I suggested earlier...
- the end of the scrolling sequences aren't smooth
- whilst the widgets are static, with sleep(2.5), they aren't updated anymore

I didn't find the correct way to smooth the scrolling and to force refresh the widgets when they are static for some seconds, I would have to dig into luma lib. The best I could do now is the following modification, calling more often 'yield x'.

Clock hands and the CPU usage graphics are updated when the widgets are static, scrolling is acceptable, Pi CPU load also (some 15% CPU at 600kHz).
```
$ diff ./carousel.py carousel.speed.py -u
--- ./carousel.py       2019-12-07 00:03:41.623544517 +0100
+++ carousel.speed.py   2019-12-08 00:44:18.204212521 +0100
@@ -16,6 +16,7 @@
 """

 import psutil
+import time

 from demo_opts import get_device
 from luma.core.virtual import viewport, snapshot
@@ -38,9 +39,13 @@
         while True:
             x = next(generator)
             if x % interval == 0:
-                for _ in range(20):
+                # Widget pause
+                for y in range(50):
+                    time.sleep(0.05)
                     yield x
             else:
+                # Scrolling speed
+                time.sleep(0.025)
                 yield x
     except StopIteration:
         pass

```

thijstriemstra, 2019-12-19:
thanks for the feedback @bk4nt. Making a pull request is super easy, simply go to https://github.com/rm-hull/luma.examples/edit/master/examples/carousel.py and 'commit changes' at the bottom of the page. This will create a new pull request.

rm-hull, 2020-10-08:
It should probably use the framerate_regulator class to govern the refresh rate rather than inserting random sleeps

closed 2020-10-08 by rm-hull