rm-hull/luma.examples #111: How to add more types of screens ?

markbirss, 2019-12-26:
How do i add for example the ST7920 oled screen ?

thijstriemstra, 2019-12-26:
Very generic question, can you be more specific?

markbirss, 2019-12-26:
sure, currently the ST7920 oled screen is unsupported, what changes are required to add support for this screen

rm-hull, 2019-12-26:
ST7920 is an LCD screen rather than OLED.

You would need to find the datasheet for the device, read it and understand how to send data to the display (whether that is via I2C, SPI or parallel bIt-banging). Then use one of the other implementations as a guide, implement a class that extends device (see https://luma-oled.readthedocs.io/en/latest/api-documentation.html) where its display() method takes a pillow image and converts the pixels to the correct commands and data for device