rm-hull/luma.examples #112: Using Material Design Icons as font

nickehallgren, 2020-01-06:
I looked at the fontawesome icons example and used that as my starting point and it worked great until I noticed that the icons I need have five characters. The code I use looks like this:

`text = '\uF02D1'`

should be the "mdi-battery-charging-high" icon but the output to the screen (SSD1306) is the "mdi-amazon" ('\uF02D'). Is this a bug or am I doing it wrong?

https://cdn.materialdesignicons.com/4.7.95/

thijstriemstra, 2020-01-09:
ping @rm-hull

rm-hull, 2020-01-09:
Working code example showing the problem would be a great help

nickehallgren, 2020-01-09:
The code is very basic.
```
font1 = ImageFont.truetype('fonts/materialdesignicons-webfont.ttf',40)

with canvas(device) as draw:
  text = '\uF02D1'
  draw.rectangle(device.bounding_box, outline="white", fill="black")
  draw.text((1, 1), text, font=font1, fill="white")
```

rm-hull, 2020-01-12:
`draw` is simply a Pillow ImageDraw instance, so the `draw.text(...)` method invoked is actually https://pillow.readthedocs.io/en/stable/reference/ImageDraw.html#PIL.ImageDraw.PIL.ImageDraw.ImageDraw.text

Hence, I suspect that Pillow and/or FreeType possibly doesn't fully support the full range of unicode characters. Before we investigate any further, what version of Pillow and FreeType are you using?

nickehallgren, 2020-01-12:
```
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PIL
>>> PIL.PILLOW_VERSION
'6.2.1'
>>>
```

How do I get the freetype version? Tried import freetype but I don't have it on my system...

rm-hull, 2020-01-13:
Ok, turns out `\u` is the 16-bit unicode hex representation, so only reads the following 4 characters. To get 32-bit unicode values, use `\U` (upper-case) instead ... so the following should work as expected:

```python
text = '\U000F02D1'
```

_I learned something new today...!_

nickehallgren, 2020-01-13:
So did I, I tried

`text = '\u000F02D1'
`
earlier but it didn't work and know I understand why. Thanks!

closed 2020-01-13 by nickehallgren