rm-hull/luma.examples #113: Multiple scripts write to single max7219

cwiggins1126, 2020-01-08:
Working on a smart clock that displays the current time and interrupts the clock display to display a message received via ifttt webhook. Periodically, I interrupt the clock to display all the days messages received.
So I have 3 scripts that could write to the device. 

Is there a "busy" function available to avoid display conflicts?
Thanks

rm-hull, 2020-01-09:
There is nothing in the luma codebase to do this (nor should there be). Your python scripts should use some system-wide mutex to prevent display conflicts. See https://stackoverflow.com/questions/6931342/system-wide-mutex-in-python-on-linux as a starter for 10

cwiggins1126, 2020-01-09:
Thanks Rich, I saw that SPI doesn't have a busy function (by design). I'll look at alternatives. I have used, a common file and an unused GPIO pin to pass busy state.