rm-hull/luma.examples #117: No module named luma.core

bennetdev, 2020-05-30:
I have followed all the required steps. I can run the examples with the python command, but if I use python3 it says "No module named 'luma.core'". I have used pip3 show luma.core and pip3 install luma.core trying to fix it but I am certain that luma.core for python3 is installed on my raspberry. Any tips?

bennetdev, 2020-05-30:
Also pip3 show luma.core shows this location:" /usr/local/lib/python3.7/dist-packages/luma.core-1.14.0-py3.7.egg" which I don't know wether its normal.

rm-hull, 2020-05-31:
Which version of raspian?
What does `python --version` show?
What does `python3 --version` show?

bennetdev, 2020-05-31:
raspbian version is the latest one (Version 10 buster)
python version 2.7.16
python3 version 3.7.3

thijstriemstra, 2020-08-11:
can you try running python3 and then 'import luma.core'?

thijstriemstra, 2020-09-04:
no feedback, closing.

closed 2020-09-04 by thijstriemstra