rm-hull/luma.examples #118: pip3 show luma.core empty

SkyperTHC, 2020-06-03:
Hi,

installs fine but module not available.
pi@Zero3:~ $ sudo pip3 install luma
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting luma
  Using cached https://files.pythonhosted.org/packages/72/a3/24551b29b61260f2b4d75cc4cc2e13d84c43605b8daa814f35a6428d65c7/luma-1.0.4-py3-none-any.whl
Requirement already satisfied: pycparser==2.18 in /usr/local/lib/python3.7/dist-packages (from luma) (2.18)
Requirement already satisfied: requests>=2.22.0 in /usr/local/lib/python3.7/dist-packages (from luma) (2.23.0)
Requirement already satisfied: PyQRCode==1.2.1 in /usr/local/lib/python3.7/dist-packages (from luma) (1.2.1)
Requirement already satisfied: python-dateutil>=2.8 in /usr/local/lib/python3.7/dist-packages (from luma) (2.8.1)
Requirement already satisfied: click>=7.0 in /usr/local/lib/python3.7/dist-packages (from luma) (7.1.2)
Requirement already satisfied: pyparsing==2.2.0 in /usr/local/lib/python3.7/dist-packages (from luma) (2.2.0)
Requirement already satisfied: docker>=4.0.1 in /usr/local/lib/python3.7/dist-packages (from luma) (4.2.1)
Requirement already satisfied: idna<3,>=2.5 in /usr/lib/python3/dist-packages (from requests>=2.22.0->luma) (2.6)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/lib/python3/dist-packages (from requests>=2.22.0->luma) (1.24.1)
Requirement already satisfied: chardet<4,>=3.0.2 in /usr/lib/python3/dist-packages (from requests>=2.22.0->luma) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests>=2.22.0->luma) (2018.8.24)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.8->luma) (1.12.0)
Requirement already satisfied: websocket-client>=0.32.0 in /usr/local/lib/python3.7/dist-packages (from docker>=4.0.1->luma) (0.57.0)
Installing collected packages: luma
**Successfully installed luma-1.0.4**

pi@Zero3:~ $ sudo pip3 show luma.core
pi@Zero3:~ $

SkyperTHC, 2020-06-03:
work around:
$ sudo apt-get install libfreetype6-dev libjpeg-dev build-essential libsdl-dev libportmidi-dev libsdl-ttf2.0-dev libsdl-mixer1.2-dev libsdl-image1.2-dev
$ sudo pip3 install luma.core
$ sudo pip3 install luma.oled
$ pip3 show luma.core
Name: luma.core
Version: 1.14.0
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python3.7/dist-packages
Requires: pyftdi, RPI.GPIO, spidev, pillow, smbus2
Required-by: luma.oled

hmm. perhaps there is a naming conflict with 'luma' package. Documentation says to install luma oled support with 'pip install luma' but that luma-package seems to be a totally unrelated package:
pip3 show luma
Name: luma
Version: 1.0.4
Summary: A CLI to interact with the Lumavate platform
Home-page: UNKNOWN
Author: Gunnar Norred
Author-email: g.norred@lumavate.com
...

thijstriemstra, 2020-06-03:
> Documentation says to install luma oled support with 'pip install luma' 

It does? Where?

SkyperTHC, 2020-06-03:
nowhere. I can not find it anymore where I saw it. I start to believe that I misread it. Sorry. 

ps. Unfortunate that pip/python does not enforce discrete and unique names for packages.  We end up with one package being called 'luma' and another one 'luma.core' and both have nothing to do with each other (e.g. luma.core is not a module of luma).

closed 2020-06-03 by SkyperTHC

rm-hull, 2020-06-03:
Agree about discrete names. We were first to publish under the luma namespace, not that really matters, but there is little to nothing we can do to prevent this sort of confusion