rm-hull/luma.examples #123: ERROR:sudo pip3 install -e

ThivagarK, 2020-09-04:
Why it is showing error although I change my python version?
pi@raspberrypi:~/max7219 $ sudo pip3 install -e

Usage:   
  pip3 install [options] <requirement specifier> [package-index-options] ...
  pip3 install [options] -r <requirements file> [package-index-options] ...
  pip3 install [options] [-e] <vcs project url> ...
  pip3 install [options] [-e] <local project path> ...
  pip3 install [options] <archive url/path> ...

-e option requires 1 argument
pi@raspberrypi:~/max7219 $

rm-hull, 2020-09-04:
As per #122, please install only supported packages and follow the installation instructions rather than outdated web tutorials.

closed 2020-09-04 by rm-hull

thijstriemstra, 2020-09-04:
use `sudo pip3 install -e .` instead (notice period at end).