rm-hull/luma.examples #135: OLED SSD1306

ruozhiertong, 2020-11-26:
My situation:
Pi 4B with2020-08-20-raspios-buster-armhf , Python3,  0.9 inch OLED with SSD1306.
When i run : python3 demo.py
There is an error:
demo.py: error: Invalid/unsupported framebuffer: /dev/fd0.
How to fix the error?
Thx.

rm-hull, 2020-11-26:
python3 demo.py -d ssd1306

rm-hull, 2020-11-26:
There was a regression in luma.core that caused this.. it should default to SSD1306, I will apply a fix now

rm-hull, 2020-11-26:
luma.core 2.0.2 released with a fix for this

closed 2020-11-26 by rm-hull

mguzelocak, 2024-09-10:
I have this problem with Pi 5. What should I do to fix the error?

when i run the file i got this error:
maze.py: error: Invalid/unsupported framebuffer: /dev/fd0

then i run it again with this command:
mcguzelocak@raspberrypi1:/home/mcguzelocak/luma.examples/examples$ python3 demo.py -d ssd1306

and I got his error:
demo.py: error: argument --display/-d: invalid choice: 'ssd1306' (choose from 'linux_framebuffer')