rm-hull/luma.examples #136: terminal.py: error: Unsupported display mode: 72 x 40

aphorise, 2020-12-08:
The code runs fine - however adjusting with the provided options for my 0.42" display that's 72x40 I get an error:

```bash
python3 terminal.py  --display ssd1306 --width 72 --height 40
# ... default help menu +
# terminal.py: error: Unsupported display mode: 72 x 40
```

thijstriemstra, 2020-12-08:
@aphorise interesting. do you have a link to the display's product or ebay page?

thijstriemstra, 2020-12-09:
Looks like something like this: https://www.buydisplay.com/i2c-white-0-42-inch-oled-display-module-72x40-arduino-raspberry-pi

That's too cute, didn't know they exist, I also ordered a few :) Looking at luma.oled, this resolution isn't supported on ssd1306 yet.

@rm-hull can you move this ticket?