rm-hull/luma.examples #137: Can't get these to display correctly - max7129 (x4)

RamblingGeekUK, 2020-12-23:
python3 bitstamp_ticker.py --interface spi --display max7219 --block-orientation -90 --width 32 --height 8

this is the output... I have tried all sorts of options, they never seem to display correctly.. what am I missing ?  

[image]

RamblingGeekUK, 2020-12-23:
python3 matrix_demo.py --block-orientation 90 --cascaded 4 --reverse-order -9 - this works, so it appears to be the --reverse-order -9, not  figured out how to do the same with these examples

RamblingGeekUK, 2020-12-23:
So Demo is working... this now makes me thing that not all these examples work on all screens... I thought the code took account for the screen.

rm-hull, 2020-12-23:
Yep, so you’ve discovered that the max7219 pcb’s can be wired up differently.
As to the luma.examples, mostly they are targeted to work with marginally bigger screens than 32x8 pixels. The main problem is getting a ttf font that works at such small pixel size: I think that is why the Bitcoin ticker doesn’t quite fit on the 8 pixels of vertical height. There are some examples in the luma.led-matrix repo that are specifically targeted at led matrices and seven segment displays

RamblingGeekUK, 2020-12-23:
Thanks for replying so soon. Have a great Christmas. 🎄

closed 2020-12-23 by RamblingGeekUK