rm-hull/luma.examples #146: How to Easily Present Strings Onto 1.12" Mono OLED (128x128, white/black) Breakout – SPI

AlexLandherr, 2021-07-27:
I'm considering purchasing this OLED display [https://thepihut.com/products/1-12-mono-oled-128x128-white-black-breakout-spi](url) for a Raspberry Pi Zero W GPS project and am wondering how to just print simple strings representing the positional data produced by the GPS receiver.

I have gone through the examples for this display and am still not sure how to print strings to it that has the information I want. I would like to know how to do this before purchasing the display in question.

thijstriemstra, 2021-07-27:
Take a look at the https://github.com/rm-hull/luma.examples/blob/master/examples/sys_info.py example. It shows how to print various diagnostic info to the screen.

AlexLandherr, 2021-07-27:
Thank you for the suggestion!

thijstriemstra, 2021-07-27:
Yes. There's a reference to the luma.oled Python library on the page you referenced, so it should be officially supported by pihut.

closed 2021-07-27 by thijstriemstra