rm-hull/luma.examples #147: Error: No module named 'luma' when using Python 3.9.6

Gun-neR, 2021-07-29:
Things worked fine after a clean install of RPi OS Buster, but after upgrading to Python 3.9.6 none of the examples will run.

```
pi@raspberrypi:~ $ python3 --version
Python 3.9.6
pi@raspberrypi:~ $ python3 /home/pi/luma.examples/examples/pi_logo.py -d ssd1331 -i spi --width 96
Traceback (most recent call last):
  File "/home/pi/luma.examples/examples/pi_logo.py", line 12, in <module>
    from demo_opts import get_device
  File "/home/pi/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error
ModuleNotFoundError: No module named 'luma'
pi@raspberrypi:~ $
```

thijstriemstra, 2021-07-29:
> but after upgrading to Python 3.9.6

How did you upgrade?

Gun-neR, 2021-07-29:
> How did you upgrade?
https://raspberrytips.com/install-latest-python-raspberry-pi/

> Try running
That just tosses out a whole list of errors... apparently something is now broken with my python versions, or at least the symbolic links, as I can't even start Thonny anymore.

However I was able to run the example by pointing directly to python3.7 (but not any other version variations) probably becasue that is what it was installed with in the first place.  If I recall, I had to use **pip3** in order to even install it.

```python3.7 /home/pi/luma.examples/examples/pi_logo.py -d ssd1331 -i spi --width 96```

So I think the issue is at my end, with a messed up python install or something with the links, and not your library.

Sorry for false alarm.

closed 2021-07-29 by thijstriemstra